boschsensortec/BSEC-Arduino-library #1: Specified folder/zip file does not contain a valid library

jttaylor86, 2018-04-10:
Hello,

there is no .ino file in this zip file. Also when I try to add the library in the Arduino  IDE I receive these messages.

Arduino: 1.8.5 (Windows 10), Board: "ESP32 Dev Module, QIO, 80MHz, 4MB (32Mb), 921600, None"

Specified folder/zip file does not contain a valid library
Specified folder/zip file does not contain a valid library
Specified folder/zip file does not contain a valid library

I would be grateful for any help regarding this.

Thanks,

Justin

DanieleSgroi, 2018-04-10:
A txt file inside each directory instructs about filling with a valid precompiled library. 

I think we could assume we still need to download the BSEC package from Bosch website.

BoschSensortec, 2018-04-10:
@jttaylor86,

Can you try again with the updated folder structure?

BoschSensortec, 2018-04-10:
@DanieleSgroi,

One needs to only copy the binaries and place it in the respective folders.

DanieleSgroi, 2018-04-10:
@BoschSensortec 

sure, it 's what I meant: copy the binaries from the bsec package.

jttaylor86, 2018-04-10:
Hi,

where can I find the arduino-builder executable? In which folder is it?

Thanks,

Justin

BoschSensortec, 2018-04-10:
Hi @jttaylor86,

The hyperlink for download is in the instruction. 
https://github.com/BoschSensortec/BSEC-Arduino-library#3-replace-the-arduino-builder

jttaylor86, 2018-04-10:
Hi,

I receive these error messages.

Arduino: 1.8.5 (Windows 10), Board: "ESP32 Dev Module, QIO, 80MHz, 4MB (32Mb), 921600, None"

...

"C:\Users\taju\AppData\Local\Temp\arduino_build_83364/arduino.ar" -lgcc -lopenssl -lbtdm_app -lfatfs -lwps -lcoexist -lwear_levelling -lhal -lnewlib -ldriver -lbootloader_support -lpp -lsmartconfig -ljsmn -lwpa -lethernet -lphy -lapp_trace -lconsole -lulp -lwpa_supplicant -lfreertos -lbt -lmicro-ecc -lcxx -lxtensa-debug-module -lmdns -lvfs -lsoc -lcore -lsdmmc -lcoap -ltcpip_adapter -lc_nano -lrtc -lspi_flash -lwpa2 -lesp32 -lapp_update -lnghttp -lspiffs -lespnow -lnvs_flash -lesp_adc_cal -llog -lexpat -lm -lc -lheap -lmbedtls -llwip -lnet80211 -lpthread -ljson  -lstdc++  -Wl,--end-group  "-LC:\Users\taju\AppData\Local\Temp\arduino_build_83364"
C:\Users\taju\AppData\Local\Temp\arduino_build_83364\libraries\BSEC-Arduino-library-master\bsec.cpp.o:(.literal._ZN4Bsec11beginCommonEv+0x0): undefined reference to `bsec_init'

C:\Users\taju\AppData\Local\Temp\arduino_build_83364\libraries\BSEC-Arduino-library-master\bsec.cpp.o:(.literal._ZN4Bsec11beginCommonEv+0x4): undefined reference to `bsec_get_version'

C:\Users\taju\AppData\Local\Temp\arduino_build_83364\libraries\BSEC-Arduino-library-master\bsec.cpp.o:(.literal._ZN4Bsec18updateSubscriptionEP21bsec_virtual_sensor_thf+0x0): undefined reference to `bsec_update_subscription'

C:\Users\taju\AppData\Local\Temp\arduino_build_83364\libraries\BSEC-Arduino-library-master\bsec.cpp.o:(.literal._ZN4Bsec15readProcessDataEx19bsec_bme_settings_t+0x8): undefined reference to `bsec_do_steps'

C:\Users\taju\AppData\Local\Temp\arduino_build_83364\libraries\BSEC-Arduino-library-master\bsec.cpp.o:(.literal._ZN4Bsec3runEv+0x0): undefined reference to `bsec_sensor_control'

C:\Users\taju\AppData\Local\Temp\arduino_build_83364\libraries\BSEC-Arduino-library-master\bsec.cpp.o: In function `Bsec::beginCommon()':

C:\Users\taju\Documents\Arduino\libraries\BSEC-Arduino-library-master\src/bsec.cpp:355: undefined reference to `bsec_init'

C:\Users\taju\AppData\Local\Temp\arduino_build_83364\libraries\BSEC-Arduino-library-master\bsec.cpp.o: In function `Bsec::getVersion()':

C:\Users\taju\Documents\Arduino\libraries\BSEC-Arduino-library-master\src/bsec.cpp:355: undefined reference to `bsec_get_version'

C:\Users\taju\AppData\Local\Temp\arduino_build_83364\libraries\BSEC-Arduino-library-master\bsec.cpp.o: In function `Bsec::updateSubscription(bsec_virtual_sensor_t*, unsigned char, float)':

C:\Users\taju\Documents\Arduino\libraries\BSEC-Arduino-library-master\src/bsec.cpp:355: undefined reference to `bsec_update_subscription'

C:\Users\taju\AppData\Local\Temp\arduino_build_83364\libraries\BSEC-Arduino-library-master\bsec.cpp.o: In function `Bsec::readProcessData(long long, bsec_bme_settings_t)':

C:\Users\taju\Documents\Arduino\libraries\BSEC-Arduino-library-master\src/bsec.cpp:291: undefined reference to `bsec_do_steps'

C:\Users\taju\AppData\Local\Temp\arduino_build_83364\libraries\BSEC-Arduino-library-master\bsec.cpp.o: In function `Bsec::run()':

C:\Users\taju\Documents\Arduino\libraries\BSEC-Arduino-library-master\src/bsec.cpp:174: undefined reference to `bsec_sensor_control'

collect2.exe: error: ld returned 1 exit status

Using library BSEC-Arduino-library-master at version 1.4.600 in folder: C:\Users\taju\Documents\Arduino\libraries\BSEC-Arduino-library-master 
Using library Wire at version 1.0 in folder: C:\Users\taju\Documents\Arduino\hardware\espressif\esp32\libraries\Wire 
Using library SPI at version 1.0 in folder: C:\Users\taju\Documents\Arduino\hardware\espressif\esp32\libraries\SPI 
exit status 1
Error compiling for board ESP32 Dev Module.

BoschSensortec, 2018-04-10:
@jttaylor86,

Yes you need to modify the platform.txt file.

jttaylor86, 2018-04-10:
Hi,

with this text for ESP32?

recipe.c.combine.pattern="{compiler.path}{compiler.c.elf.cmd}"
{compiler.c.elf.flags}  -o "{build.path}/{build.project_name}.elf"
-Wl,--start-group {object_files} "{build.path}/arduino.ar"
{compiler.c.elf.libs} {compiler.c.elf.extra_flags} -Wl,--end-group
"-L{build.path}"

Thanks,

Justin

BoschSensortec, 2018-04-10:
Hi @jttaylor86,

Yes.

BoschSensortec, 2018-04-10:
This should replace the arduino-builder.exe in your Arduino installation.

jttaylor86, 2018-04-10:
Hi,

I replaced the arduino-builder and now I receive these errors.

Archiving built core (caching) in:
C:\Users\taju\AppData\Local\Temp\arduino_cache_26638\core\core_espressif_esp32_esp32_FlashMode_qio,FlashFreq_80,FlashSize_4M,UploadSpeed_921600,DebugLevel_none_e5353ec466995ef7300ad6dbca045078.a
Linking everything together...
"C:\Users\taju\Documents\Arduino\hardware\espressif\esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc"
-nostdlib
"-LC:\Users\taju\Documents\Arduino\hardware\espressif\esp32/tools/sdk/lib"
"-LC:\Users\taju\Documents\Arduino\hardware\espressif\esp32/tools/sdk/ld"
-T esp32_out.ld -T esp32.common.ld -T esp32.rom.ld -T esp32.peripherals.ld
-T esp32.rom.spiram_incompatible_fns.ld -u ld_include_panic_highint_hdl -u
call_user_start_cpu0 -Wl,--gc-sections -Wl,-static
-Wl,--undefined=uxTopUsedPriority  -u __cxa_guard_dummy -u
__cxx_fatal_exception  -o
"C:\Users\taju\AppData\Local\Temp\arduino_build_417411/basic_without_leds.ino.elf"
-Wl,--start-group
"C:\Users\taju\AppData\Local\Temp\arduino_build_417411\sketch\basic_without_leds.ino.cpp.o"
"C:\Users\taju\AppData\Local\Temp\arduino_build_417411\libraries\BSEC-Arduino-library-master\bsec.cpp.o"
"C:\Users\taju\AppData\Local\Temp\arduino_build_417411\libraries\BSEC-Arduino-library-master\bme680\bme680.c.o"
"C:\Users\taju\AppData\Local\Temp\arduino_build_417411\libraries\Wire\Wire.cpp.o"
"C:\Users\taju\AppData\Local\Temp\arduino_build_417411\libraries\SPI\SPI.cpp.o"
"C:\Users\taju\AppData\Local\Temp\arduino_build_417411/arduino.ar" -lgcc
-lopenssl -lbtdm_app -lfatfs -lwps -lcoexist -lwear_levelling -lhal
-lnewlib -ldriver -lbootloader_support -lpp -lsmartconfig -ljsmn -lwpa
-lethernet -lphy -lapp_trace -lconsole -lulp -lwpa_supplicant -lfreertos
-lbt -lmicro-ecc -lcxx -lxtensa-debug-module -lmdns -lvfs -lsoc -lcore
-lsdmmc -lcoap -ltcpip_adapter -lc_nano -lrtc -lspi_flash -lwpa2 -lesp32
-lapp_update -lnghttp -lspiffs -lespnow -lnvs_flash -lesp_adc_cal -llog
-lexpat -lm -lc -lheap -lmbedtls -llwip -lnet80211 -lpthread -ljson
-lstdc++
"-LC:\Users\taju\Documents\Arduino\libraries\BSEC-Arduino-library-master\src\esp32"
-lalgobsec  -Wl,--end-group
"-LC:\Users\taju\AppData\Local\Temp\arduino_build_417411"
c:/users/taju/documents/arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld.exe:
cannot find -lalgobsec

collect2.exe: error: ld returned 1 exit status

Using library BSEC-Arduino-library-master at version 1.4.600 in folder:
C:\Users\taju\Documents\Arduino\libraries\BSEC-Arduino-library-master
Using library Wire at version 1.0 in folder:
C:\Users\taju\Documents\Arduino\hardware\espressif\esp32\libraries\Wire
Using library SPI at version 1.0 in folder:
C:\Users\taju\Documents\Arduino\hardware\espressif\esp32\libraries\SPI
exit status 1
Error compiling for board ESP32 Dev Module.

I would be grateful for any help.

Thanks,

Justin

BoschSensortec, 2018-04-10:
Hi @jttaylor86,

Download the library zip on our [website](https://www.bosch-sensortec.com/en/bst/products/all_products/bsec),  
Copy the file `libalgobsec.a` from `algo/bin/Normal_version/esp32` in the zip archive to `C:\Users\taju\Documents\Arduino\libraries\BSEC-Arduino-library-master\src\esp32` in your PC.

jttaylor86, 2018-04-10:
Hi,

thanks. Now it compiles.

I would recommend including this last step in the  "Instructions for using
the BSEC Arduino Library in Arduino 1.8.5". Maybe it is perfectly clear for
others but it wasn't for me.

Thank you very much for your help.

jttaylor86, 2018-04-10:
Hi,

How many does the sensor take until it is ready?

On the serial monitor I receive this info and nothing else.

ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13076
entry 0x40078ad0

BSEC library version 1.4.6.0
Timestamp [ms], raw temperature [°C], pressure [hPa], raw relative humidity
[%], gas [Ohm], IAQ,

Thanks,

Justin

2018-04-10 15:47 GMT+02:00 Justin Taylor <jttaylor86@gmail.com>:

jttaylor86, 2018-04-11:
Hello,

does the sensor output IAQ: 25 as standard for the first X reading or X
times with the basic sketch?

Thanks,

Justin

2018-04-10 16:36 GMT+02:00 Justin Taylor <jttaylor86@gmail.com>:

jttaylor86, 2018-04-11:
Hi,

in the void loop() I have a call to another function to write the IAQ Value
among others to Thingspeak however the IAQ Value never updates although the
other values (humidity, temperature) are updated.

What could be the reason for this?

Thanks

2018-04-11 9:44 GMT+02:00 Justin Taylor <jttaylor86@gmail.com>:

BoschSensortec, 2018-04-11:
Hi @jttaylor86,

The IAQ calibration takes 5 minutes for the first run. After the calibration you will receive IAQ values. Monitoring IAQ Accuracy is typically useful, IAQ values can be used if IAQ Accuracy is non-zero.

jttaylor86, 2018-04-11:
Hi,

would it be possible to get your email address so I don`t have to clutter
up the forum? I have a few more questions.

Thanks.

jttaylor86, 2018-04-11:
> Hi @jttaylor86 <https://github.com/jttaylor86>,
>
> You can request for a contact person using our contact form
> <https://www.bosch-sensortec.com/bst/contact/contact_form/contact_form>.

Hi,

before I done. Could you answer this please?

in the void loop() I have a call to another function to write the IAQ Value
among others to Thingspeak, however, the IAQ Value never updates although
the other values (humidity, temperature) are updated.

What could be the reason for this?

Thanks

jttaylor86, 2018-04-11:
> Hi @jttaylor86 <https://github.com/jttaylor86>,
>
> Values get updated only after 5 minutes as mentioned above.

Hi,

after the initial 5 minutes "warm-up" the values are updated each time
around the loop. Try it out.

I exposed the sensor to ethanol and the IAQ value changed immediately.
However, when I call away from main the value doesn't update.

Thanks,

Justin

BoschSensortec, 2018-04-11:
Hi @jttaylor86,

Can you elaborate a bit on,

> However, when I call away from main the value doesn't update.

jttaylor86, 2018-04-11:
Hi,

in the basic sketch the loop looks like this apart from thingSpeak();

void loop(void)
{
  if (iaqSensor.run()) { // If new data is available
    output = String(millis());
    output += ", " + String(iaqSensor.rawTemperature);
    output += ", " + String(iaqSensor.pressure);
    output += ", " + String(iaqSensor.rawHumidity);
    output += ", " + String(iaqSensor.gasResistance);
    output += ", " + String(iaqSensor.iaqEstimate);
    output += ", " + String(iaqSensor.iaqAccuracy);
    output += ", " + String(iaqSensor.temperature);
    output += ", " + String(iaqSensor.humidity);
    Serial.println(output);
  } else {
    checkIaqSensorStatus();
  }
*thingSpeak();*
}

With thingSpeak(); removed the  iaqSensor.iaqEstimate updates, however,
when I include thingSpeak(); temperature, pressure, rawHumidity,
gasResistance, temperature, humidity values are updated each time around
the loop and iaqEstimate isn't.

Why is this?

Thanks.

BoschSensortec, 2018-04-11:
@jttaylor86,

Can you add the output of the Serial terminal? It could be because the thingSpeak callback generates timing violations, and hence the IAQ estimate is not calculated due to possibly high inaccuracies.

jttaylor86, 2018-04-11:
The output of  output += ", " + String(iaqSensor.iaqEstimate) stays the
same (ie not updated).

Would it be possible for someone at Sensortec check this out? Not
specifically regarding thingSpeak but generally what happens with a call to
another function regards the timing.

My application is to use the BME680 as an IoT Sensor, and if I can't write
to thingSpeak or equivalent it isn't much use.

Thanks

oleksid, 2018-04-13:
Hello.
It seems that I have the same or similar error to @jttaylor86.
And I already did all the steps mentioned in BSEC Arduino Library in Arduino 1.8.5 and Integration Guide
Bosch Software Environmental Cluster (BSEC), but it says that it is impossible to compile to Adafruit Feather HUZZAH ESP8266.

`Archiving built core (caching) in: C:\Users\user1\AppData\Local\Temp\arduino_cache_379582\core\core_esp8266com_esp8266_huzzah_CpuFrequency_80,VTable_flash,FlashSize_4M1M,LwIPVariant_v2mss536,Debug_Disabled,DebugLevel_None____,FlashErase_none,UploadSpeed_115200_59f27b87b1649ba33697614d4ec8334b.a
Linking everything together...
"C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc" -CC -E -P -DVTABLES_IN_FLASH "C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266/tools/sdk/ld/eagle.app.v6.common.ld.h" -o "C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266/tools/sdk/ld/eagle.app.v6.common.ld"
"C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc" -Wl,-Map "-Wl,C:\Users\user1\AppData\Local\Temp\arduino_build_868996/basic.ino.map" -g -w -Os -nostdlib -Wl,--no-check-sections -u call_user_start -u _printf_float -u _scanf_float -Wl,-static "-LC:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266/tools/sdk/lib" "-LC:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266/tools/sdk/ld" "-LC:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266/tools/sdk/libc/xtensa-lx106-elf/lib" "-Teagle.flash.4m1m.ld" -Wl,--gc-sections -Wl,-wrap,system_restart_local -Wl,-wrap,spi_flash_read "-LC:\Users\user1\Documents\Arduino\libraries\BSEC-Arduino-library-master\src\esp8266" -lalgobsec  -o "C:\Users\user1\AppData\Local\Temp\arduino_build_868996/basic.ino.elf" -Wl,--start-group "C:\Users\user1\AppData\Local\Temp\arduino_build_868996\sketch\basic.ino.cpp.o" "C:\Users\user1\AppData\Local\Temp\arduino_build_868996\libraries\BSEC-Arduino-library-master\bsec.cpp.o" "C:\Users\user1\AppData\Local\Temp\arduino_build_868996\libraries\BSEC-Arduino-library-master\bme680\bme680.c.o" "C:\Users\user1\AppData\Local\Temp\arduino_build_868996\libraries\Wire\Wire.cpp.o" "C:\Users\user1\AppData\Local\Temp\arduino_build_868996\libraries\SPI\SPI.cpp.o" "C:\Users\user1\AppData\Local\Temp\arduino_build_868996/arduino.ar" -lhal -lphy -lpp -lnet80211 -llwip2 -lwpa -lcrypto -lmain -lwps -laxtls -lespnow -lsmartconfig -lairkiss -lwpa2 -lstdc++ -lm -lc -lgcc -Wl,--end-group  "-LC:\Users\user1\AppData\Local\Temp\arduino_build_868996"
C:\Users\user1\AppData\Local\Temp\arduino_build_868996\libraries\BSEC-Arduino-library-master\bsec.cpp.o: In function `Bsec::i2cRead(unsigned char, unsigned char, unsigned char*, unsigned short)':

C:\Users\user1\Documents\Arduino\libraries\BSEC-Arduino-library-master\src/bsec.cpp:355: undefined reference to `bsec_init'

C:\Users\user1\Documents\Arduino\libraries\BSEC-Arduino-library-master\src/bsec.cpp:355: undefined reference to `bsec_get_version'

C:\Users\user1\AppData\Local\Temp\arduino_build_868996\libraries\BSEC-Arduino-library-master\bsec.cpp.o: In function `Bsec::beginCommon()':

C:\Users\user1\Documents\Arduino\libraries\BSEC-Arduino-library-master\src/bsec.cpp:355: undefined reference to `bsec_init'

C:\Users\user1\Documents\Arduino\libraries\BSEC-Arduino-library-master\src/bsec.cpp:355: undefined reference to `bsec_get_version'

C:\Users\user1\AppData\Local\Temp\arduino_build_868996\libraries\BSEC-Arduino-library-master\bsec.cpp.o: In function `Bsec::begin(unsigned char, TwoWire&)':

C:\Users\user1\Documents\Arduino\libraries\BSEC-Arduino-library-master\src/bsec.cpp:355: undefined reference to `bsec_update_subscription'

C:\Users\user1\AppData\Local\Temp\arduino_build_868996\libraries\BSEC-Arduino-library-master\bsec.cpp.o: In function `Bsec::updateSubscription(bsec_virtual_sensor_t*, unsigned char, float)':

C:\Users\user1\Documents\Arduino\libraries\BSEC-Arduino-library-master\src/bsec.cpp:355: undefined reference to `bsec_update_subscription'

C:\Users\user1\AppData\Local\Temp\arduino_build_868996\libraries\BSEC-Arduino-library-master\bsec.cpp.o: In function `Bsec::Bsec()':

C:\Users\user1\Documents\Arduino\libraries\BSEC-Arduino-library-master\src/bsec.cpp:69: undefined reference to `bsec_do_steps'

C:\Users\user1\AppData\Local\Temp\arduino_build_868996\libraries\BSEC-Arduino-library-master\bsec.cpp.o: In function `Bsec::readProcessData(long long, bsec_bme_settings_t)':

C:\Users\user1\Documents\Arduino\libraries\BSEC-Arduino-library-master\src/bsec.cpp:287: undefined reference to `bsec_do_steps'

C:\Users\user1\AppData\Local\Temp\arduino_build_868996\libraries\BSEC-Arduino-library-master\bsec.cpp.o: In function `Bsec::getTimeMs()':

C:\Users\user1\Documents\Arduino\libraries\BSEC-Arduino-library-master\src/bsec.cpp:387: undefined reference to `bsec_sensor_control'

C:\Users\user1\AppData\Local\Temp\arduino_build_868996\libraries\BSEC-Arduino-library-master\bsec.cpp.o: In function `Bsec::run()':

C:\Users\user1\Documents\Arduino\libraries\BSEC-Arduino-library-master\src/bsec.cpp:163: undefined reference to `bsec_sensor_control'

collect2.exe: error: ld returned 1 exit status

Using library BSEC-Arduino-library-master at version 1.4.600 in folder: C:\Users\user1\Documents\Arduino\libraries\BSEC-Arduino-library-master 
Using library Wire at version 1.0 in folder: C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\Wire 
Using library SPI at version 1.0 in folder: C:\Program Files (x86)\Arduino\hardware\esp8266com\esp8266\libraries\SPI 
exit status 1
Error compiling for board Adafruit Feather HUZZAH ESP8266.
`

oleksid, 2018-04-20:
@jttaylor86 

Hello.

Are you using ThingSpeak to send the data from BME680?

Thanks,
Oleksandra S.

axelbusch, 2018-05-01:
Hi,
I replaced the builder and modified the platform.txt as described in Justins post. However, I get the following errors
`Arduino: 1.8.4 (Windows 10), Board: "DOIT ESP32 DEVKIT V1, 80MHz, 512000, Debug"

Build-Optionen wurden verändert, alles wird neu kompiliert
Archiving built core (caching) in: C:\Users\Konto\AppData\Local\Temp\arduino_cache_102738\core\core_espressif_esp32_esp32doit-devkit-v1_FlashFreq_80,UploadSpeed_512000,DebugLevel_debug_7c1566cab1486550fedc44c7633542a3.a
libraries\BSEC-Arduino-library-master\bsec.cpp.o:(.literal._ZN4Bsec11beginCommonEv+0x0): undefined reference to `bsec_init'

libraries\BSEC-Arduino-library-master\bsec.cpp.o:(.literal._ZN4Bsec11beginCommonEv+0x4): undefined reference to `bsec_get_version'

libraries\BSEC-Arduino-library-master\bsec.cpp.o:(.literal._ZN4Bsec18updateSubscriptionEP21bsec_virtual_sensor_thf+0x0): undefined reference to `bsec_update_subscription'

libraries\BSEC-Arduino-library-master\bsec.cpp.o:(.literal._ZN4Bsec15readProcessDataEx19bsec_bme_settings_t+0x8): undefined reference to `bsec_do_steps'

libraries\BSEC-Arduino-library-master\bsec.cpp.o:(.literal._ZN4Bsec3runEv+0x0): undefined reference to `bsec_sensor_control'

libraries\BSEC-Arduino-library-master\bsec.cpp.o: In function `Bsec::beginCommon()':

C:\Users\Konto\Documents\Arduino\libraries\BSEC-Arduino-library-master\src/bsec.cpp:355: undefined reference to `bsec_init'

libraries\BSEC-Arduino-library-master\bsec.cpp.o: In function `Bsec::getVersion()':

C:\Users\Konto\Documents\Arduino\libraries\BSEC-Arduino-library-master\src/bsec.cpp:355: undefined reference to `bsec_get_version'

libraries\BSEC-Arduino-library-master\bsec.cpp.o: In function `Bsec::updateSubscription(bsec_virtual_sensor_t*, unsigned char, float)':

C:\Users\Konto\Documents\Arduino\libraries\BSEC-Arduino-library-master\src/bsec.cpp:355: undefined reference to `bsec_update_subscription'

libraries\BSEC-Arduino-library-master\bsec.cpp.o: In function `Bsec::readProcessData(long long, bsec_bme_settings_t)':

C:\Users\Konto\Documents\Arduino\libraries\BSEC-Arduino-library-master\src/bsec.cpp:291: undefined reference to `bsec_do_steps'

libraries\BSEC-Arduino-library-master\bsec.cpp.o: In function `Bsec::run()':

C:\Users\Konto\Documents\Arduino\libraries\BSEC-Arduino-library-master\src/bsec.cpp:174: undefined reference to `bsec_sensor_control'

collect2.exe: error: ld returned 1 exit status

exit status 1
`

oleksid, 2018-05-21:
Hello @jttaylor86 

I have the same problem too.

I want to send BME680 data to ThingSpeak and I can do it but the IAQ accuracy is always 0.

Have you managed to solve this problem?

Thank you.