boschsensortec/BSEC-Arduino-library #2: Can't compile on Arduino MEGA 2560

DanieleSgroi, 2018-04-10:
Downloded and patched as instructed.

Skipped step 4, assumingit does not apply anymore to Arduino MEGA 2560...

Added libalgobsec.a binary file from BSEC_1.4.6.0_Generic_Release_20180323.zip package (still with -fshort_enums issue).

The provided examples does not recognize any of the library calls: each function return a "undefined reference" compiler error.

Have you tried these examples before posting?

Cheers

BoschSensortec, 2018-04-10:
Hi @DanieleSgroi,

Step 4 applies for all platforms. Without this step, the library flags will be discarded by the linker.

BoschSensortec, 2018-04-10:
@DanieleSgroi,

In your opinion, would this serve as a replacement? 

The `arduino-builder-PR219` now has the ability to select the appropriate pre-compiled library from the library folder. This avoids the need to add the linker flag `-libalgobsec` manually to the platform.txt file and copying the pre-compiled library `libalgobsec.a` to a core specific directory. Instead the binary would be part of the Arduino library and the linker flag passed to `compiler.c.elf.extra_flags`.

Replace the arduino-builder executable from the Arduino folder with the appropriate file based on your OS from this link [arduino-builder-219.zip](http://downloads.arduino.cc/PR/arduino-builder/arduino-builder-219.zip).

DanieleSgroi, 2018-04-10:
@BoschSensortec 

It was perfectly clear the need of the precompiled library, to be copied now into the library folder (and not anymore by tweaking the core compiler structure).

From the above proposed amended text, I still understand that I don't need anymore to manually edit the platform.txt file, since the new arduino-builder-PR219 will take care of finding the lib.

This text is still in contrast with your answer above, where you say that step 4 (manually editing the platform.txt file) does apply to any platform including Arduino MEGA (for which you don't provide any editing example).

thank you

DanieleSgroi, 2018-04-10:
You can amend the section 4 adding this example for Arduino platform (which gives it's name to the repository, at least):

Original lines 67, 68:
## Combine gc-sections, archives, and objects
recipe.c.combine.pattern="{compiler.path}{compiler.c.elf.cmd}" {compiler.c.elf.flags} -mmcu={build.mcu} {compiler.c.elf.extra_flags} -o "{build.path}/{build.project_name}.elf" {object_files} "{build.path}/{archive_file}" "-L{build.path}" -lm

Should be:
## Combine gc-sections, archives, and objects
recipe.c.combine.pattern="{compiler.path}{compiler.c.elf.cmd}" {compiler.c.elf.flags} -mmcu={build.mcu} -o "{build.path}/{build.project_name}.elf" {object_files} "{build.path}/{archive_file}" {compiler.c.elf.extra_flags} "-L{build.path}" -lm

closed 2018-04-10 by DanieleSgroi

jttaylor86, 2018-07-17:
I am confused. Do I have to change the platform text file for ESP32 or not?

If I don't change it, I can't compile the code. If I do change it, I am not following the instructions from Bosch.

What do I need to do to make the bsec al compile with the ESP32.

@BoschSensortec could you make the instructions more clear please?

Thanks,
Justin