boschsensortec/BSEC-Arduino-library #3: BME680 and ESP8266 connection

oleksid, 2018-04-10:
Hello.

The connection between BME680 and ESP8266 have to be I2C or SPI to use the BSEC software?
What it is necessary to write in the program to define the connections?

Thanks,
Oleksandra.

BoschSensortec, 2018-04-10:
Hi @oleksid,

If you are using the Arduino IDE to program the ESP8266, can you share some information on your hardware setup. Also, instructions on how to use the library are provided in the README.

oleksid, 2018-04-10:
I am using BME680 Shuttle Board from Bosch Sensortec and Adafruit Feather HUZZAH with ESP8266. 
The connection that I am trying to use is I2C.

BoschSensortec, 2018-04-10:
@oleksid

| Shuttle pin number (name) | HUZZAH name | Comments |
|------------------------------|------------------|-------------|
| 1 (VDD) | 3v3 | |
| 2 (VDDIO) | 3v3 | |
| 3 (GND) | GND | |
| 4 (SDO) | GND | Address select. 3v3 for 0x77 |
| 5 (SDI) | SDA | |
| 6 (SCK) | SCL | |
| 7 (CS) | 3v3 | |

oleksid, 2018-04-11:
I made this connection and I'm using the basic.ino example.
I followed all the steps from BSEC Integration Guideline and BSEC-Arduino-library but I don't have output in serial monitor.

BoschSensortec, 2018-04-11:
Hi @oleksid,

Is the LED blinking rapidly?

oleksid, 2018-04-11:
Yes, during the uploading.

oleksid, 2018-04-11:
It was the problem! Thanks.

Why the iaq Accuracy is 0?

BoschSensortec, 2018-04-11:
@oleksid,

The accuracy is 0 for the first 5 minutes because the algorithm is calibrating.

oleksid, 2018-04-11:
I already notice this observing the serial monitor!
Is it possible to implement SPI connection and use basic.ino?

BoschSensortec, 2018-04-11:
@oleksid,

Yes it is. Change the begin function like below,

```c
iaqSensor.begin(10, SPI); /* Digital pin used as chip select */
```

oleksid, 2018-04-11:
@BoschSensortec 
Ok, thank you a lot for all the information!

closed 2018-04-11 by BoschSensortec