boschsensortec/BSEC-Arduino-library #4: 'SPIClass' does not name a type   static SPIClass *spiObj;

r0ehre, 2018-04-10:
I get the following error when trying to compile the example.

```
WARNING: library BSEC-Arduino-library-master claims to run on (samd, sam, esp8266, nrf52, esp32) architecture(s) and may be incompatible with your current board which runs on (nRF52832) architecture(s).
In file included from /Users/tobiasroddiger/Documents/Arduino/_/_.ino:1:0:
/Users/tobiasroddiger/Documents/Arduino/libraries/BSEC-Arduino-library-master/src/bsec.h:72:9: error: 'SPIClass' does not name a type
  static SPIClass *spiObj;
         ^
In file included from /Users/tobiasroddiger/Documents/Arduino/_/_.ino:1:0:
/Users/tobiasroddiger/Documents/Arduino/libraries/BSEC-Arduino-library-master/src/bsec.h:102:33: error: 'SPIClass' has not been declared
  void begin(uint8_t chipSelect, SPIClass &spi);
```

BoschSensortec, 2018-04-11:
@r0ehre,

It seems like it you are using a different implementation of the Arduino HAL. Can you link it here?

Also, the platform.txt combine recipe should look like
```
## Combine gc-sections, archives, and objects
recipe.c.combine.pattern="{compiler.path}{compiler.c.elf.cmd}" {compiler.c.elf.flags}  -mcpu={build.mcu} -mthumb "-T{build.variant.path}/{build.ldscript}" {compiler.ldflags} "-Wl,-Map,{build.path}/{build.project_name}.map" -o "{build.path}/{build.project_name}.elf" {build.ld_include} "-L{build.path}" -lm -lgcc -Wl,--cref -Wl,--check-sections -Wl,--gc-sections -Wl,--warn-common -Wl,--warn-unresolved-symbols -Wl,--start-group {object_files} "{build.path}/{archive_file}" -lc -lnosys {compiler.c.elf.extra_flags} -Wl,--end-group
```

r0ehre, 2018-04-11:
Thanks for the quick response - your community work is amazing! In the Arduino>packages>RedBear>hardware>nRF52832>0.0.2>libraries folder I checked for SPI and it turned out it is a custom implementation called SPI_Master. This is the library.properties file:

```
name=SPI
version=1.0
author=RedBear
maintainer=RedBear
sentence=Enables the communication with devices that use the Serial Peripheral Interface (SPI) Bus. For RedBear nRF5x.
paragraph=
category=Communication
url=https://github.com/redbear/nRF5x
architectures=
```

I changed the BSEC.h file to include the correct header file called `SPI_Master.h`. The implementation does not support SPISettings therefore I also changed the following part of the bsec.cpp implementation:
```cpp
/**
 * @brief Callback function for reading and writing registers over SPI
 */
int8_t Bsec::spiTransfer(uint8_t devId, uint8_t regAddr, uint8_t *regData, uint16_t length)
{
	int8_t rslt = 0;
	if(Bsec::spiObj) {
		Bsec::spiObj->setFrequency(4000000);
		Bsec::spiObj->setFrequency(MSBFIRST);
		Bsec::spiObj->setSPIMode(SPI_MODE0);
		
		digitalWrite(devId, LOW);

		Bsec::spiObj->transfer(regAddr); // Write the register address, ignore the return
		for (uint16_t i = 0; i < length; i++)
			regData[i] = Bsec::spiObj->transfer(regData[i]);

		digitalWrite(devId, HIGH);
		Bsec::spiObj->endTransaction();
	} else {
		rslt = -1;
	}

	return rslt;;
}
```

Unfortunately I'll be out of office till next week and will let you know if I'm successful with this code by then. Thanks again!

r0ehre, 2018-04-18:
Just wanted to share with you that everything is working now. Thanks for the great community support!

closed 2018-04-24 by BoschSensortec