boschsensortec/BSEC-Arduino-library #5: Sampling rate configruation not understood - how to achive sampling intervals other than 3sec oder 300sec ?

janzieg, 2018-04-13:
Hi @BoschSensortec,

first of all thanks for that Arduino-library! For me it works good so far...
The only issue I´m having right now is about sampling rate.

I want to sample each 60 seconds within the arduino loop function. This also means I _cannot_ sample faster (as other operations take place before that blocks the thing).

I started to configure `BSEC_SAMPLE_RATE_LP`. The problem with that was, that iaqAccuracy stays 0 if I sample every 60 seconds. When reading your spec it sais that the intverval between two measurments must not exceed 3 (or 4.5) seconds in that mode. Maybe that´s the explanation for iaqAccuracy staying zero.
So I switched to `BSEC_SAMPLE_RATE_ULP`. No everything works and iaqAccuracy rises to 3, but: I only get new data each 300 seconds now. So 4 of 5 measurements don´t get new data when sampling every 60 seconds.

My Question:
Is it possible to configure a sample rate between 3 seconds and 300 seconds ?

Does this require a special bsec configuration set via `bsec.setConfig(...)` and which samplingRate should I pass to `updateSubscription()` in that case (how does that play together actually) ?

I also tried to solve this with ULP_PLUS mode. Therefore I "triggered" an ULP_PLUS measurement like shown in your examples by just calling the ulp plus trigger function within my loop to achieve "inbetween" measurments. But this only gave me the warning status `BSEC_W_SC_MODEXCEEDULPTIMELIMIT` (ULP plus is not allowed because an ULP measurement just took or will take place) when trying to sample after this setting. And the frequency of getting new data stays about 300 seconds.

Can you give me an advice here ?
Thanks!

jttaylor86, 2018-04-13:
@janzieg I am interested in this too. Maybe this is why when I after an IAQ measurement in loop() that when I want to write the value to thingSpeak that the subsequent values of the IAQ are not updated.

Please let me know if you find a solution.

Thanks

abhinavbhargava, 2018-04-26:
I need to samples at 5 different rates and if I try to use ON_DEMAND my IAQ Accuracy falls to 0

janzieg, 2018-05-02:
Hi @BoschSensortec,

In short my code looks like this concerning the trigger function Integration:

```
void loop()
{
        …
        // check for new data every 60 seconds
        if(waitedFor60Secs)
        {
            if(bme680.run())
            {
                // hey, we got data :)
                // but: we only get data each 300 secs because of SEC_SAMPLE_RATE_ULP setting
                …
            }
            else
            {
                // no new data yet…

                // just to be sure if sensor state is ok
                checkSensorStatus();

                // let´s try to trigger ulp plus to enforce getting new ndata earlier…?
                // but: this just leads to warning status BSEC_W_SC_MODEXCEEDULPTIMELIMIT and no extra data
                triggerULPPlus();
            }
        }
        ...
}

void triggerULPPlus()
{
        bsec_virtual_sensor_t sensorList[1] = {
		BSEC_OUTPUT_IAQ_ESTIMATE,
	};

	bme680.updateSubscription(sensorList, 1, BSEC_SAMPLE_RATE_ULP_MEASUREMENT_ON_DEMAND);
	checkSensorStatus();
}
```
 
So what am I doing wrong here ?

BoschSensortec, 2018-05-03:
Hi @janzieg,

Can you try to change the implementation to,

```c
 …
       if(bme680.run())
       {
           // hey, we got data :)
           // but: we only get data each 300 secs because of SEC_SAMPLE_RATE_ULP setting
           …
       }
       else
       {
                // no new data yet…

                // just to be sure if sensor state is ok
                checkSensorStatus();
       }
       // check for new data every 60 seconds
       if(waitedFor60Secs)
       {
                // let´s try to trigger ulp plus to enforce getting new ndata earlier…?
                // but: this just leads to warning status BSEC_W_SC_MODEXCEEDULPTIMELIMIT and no extra data
                triggerULPPlus();
       }

        ...
```

r0ehre, 2018-05-03:
BSEC_SAMPLE_RATE_ULP_MEASUREMENT_ON_DEMAND also not working for me

janzieg, 2018-05-13:
@BoschSensortec Thanks for you advice. As I (can) only call ``bme680.run()`` every 60 seconds in my code, ``triggerULPPlus()`` is also only called at least every 60 seconds (if no new data was received). 

So it´s more like this:

```

void loop()
{
    // do other stuff

    if(waitedFor60Secs)
    {
        sampling();
    }
}
void sampling()
{
    if(bme680.run())
    {
        // hey, we got data :)
        // but: we only get data each 300 secs because of SEC_SAMPLE_RATE_ULP setting
        …
    }
    else
    {
        // no new data yet…

        // just to be sure if sensor state is ok
        checkSensorStatus();

        // let´s try to trigger ulp plus to enforce getting new ndata earlier…?
        // but: this just leads to warning status BSEC_W_SC_MODEXCEEDULPTIMELIMIT and no extra data
        triggerULPPlus();
    }
}
```

It tried several things to move ``triggerULPPlus()`` to another place - but it does not change the warning behavior nor forces getting new data faster. 

I would really appreciate to have more flexibility concerning sample rate interval configuration.

michapr, 2018-06-02:
If I understand right - the sensor need in this mode the commands in loop - then you can using an interrupt a call to ulp_plus_button_press(); (from sample).
This is working fine in sample code, even if you are using a software interrupt (timer) - but you will get the results in loop only at defined time in timer.

If you have your own loop() for any other stuff - it may not work more.
I have tested this -- and found one (maybe not the best) solution, what is working fine here.

You make one timer (software interrupt) for all the iaq procedure.
Within this timer you define, how often you need the value from sensor and make the call.
As sample I set the timer to 100ms - and after 600 timer calls (=60 seconds) will get a report.
(about best / max. timer interval not sure in tis time - if too large will come a warning)

Maybe this construct can help you to solve the problem in your solution.
-> And, using this way (interrupt) you can define any time for a request.
```
#include <Ticker.h>
...
Ticker iaq_caller;
int my_interval=0;
....
void setup(void)
{
...
  iaq_caller.attach(0.1,iaq_call);   // call the main procedure every 0.1 seconds 
...
}

void loop(void)
{
[your code only]
}

void iaq_call(){
  my_interval++;
  if(my_interval==600) {          // 600 for 60 seconds new value
    ulp_plus_button_press();
    my_interval=0;
  }
  if (iaqSensor.run()) { // If new data is available
    output = String(millis());
    output += ", " + String(iaqSensor.rawTemperature);
    output += ", " + String(iaqSensor.pressure);
    output += ", " + String(iaqSensor.rawHumidity);
    output += ", " + String(iaqSensor.gasResistance);
    output += ", " + String(iaqSensor.iaqEstimate);
    output += ", " + String(iaqSensor.iaqAccuracy);
    output += ", " + String(iaqSensor.temperature);
    output += ", " + String(iaqSensor.humidity);
    Serial.println(output);
    updateState();
  } else {
    checkIaqSensorStatus();
  }  
}
```

Tell me if I'm wrong - or if it help.

Best regards
Michael

janzieg, 2018-07-09:
@michapr thanks for your advice using a software Interrupt / ticker.
Unfortunately this leads to the same problem: whenever requesting ulp_plus via Ticker I get BSEC_W_SC_MODEXCEEDULPTIMELIMIT and no data - like before.

So no solution for this yet...

Argon-, 2018-08-22:
Sorry for highjacking this issue but the people who have experienced this problem (or @BoschSensortec ) probably have exactly the answer I'm looking for: does the IAQ accuracy drop to 0 even when a single measurement deadline is missed (among several deadlines that were met)?

E.g. in my case I can make a 3s sampling work, however, due to network issues I cannot guarantee it 100%. In tests I found that a single deadline might be missed every other minutes or so (due to networking). Is this enough to prevent me from getting good/accurate IAQ readings // will each of these hiccups directly reset the IAQ accuracy to zero, despite 99% of the deadlines being met?

aduecho, 2018-09-26:
@janzieg hey, I get the same problem as yours recently, after a lot  of trails, now I found how to make the ulp+ works,  the key point is that your only can call the ulp+ more than 60sec later after last measurement finished, once you call it less than 60s, you will get the error as you post. and the ulp+ seem never works again.  And another thing which is also important is you Must load the configuration with 3s not 300s. it is a quite confusing name with Lp(3S) and ULP(300S).  you need to load a 3s config but set the mode to ULP. in the reference manual there is a short description for that, but after a lot of trails I get the meaning of it:sweat_smile:

janzieg, 2018-09-27:
Hi @aduecho,

thanks for clearify! The 60sec thing, I was aware of when looking at my code. But I didn´t try changing to LP(3s) when using ulp+ - as i thought then i MUST measure every 3 seconds which was not possible then.

Meanwhile I changed my whole program to be able to perform a 3 seconds measurement instead of 60 seconds. I could achieve that all other tasks to be performed beside sampling are within that range (most of the time). As my device is not required to be super low power it doesn´t matter - otherwise ULP measurment would still be the needed setting.

aduecho, 2018-09-27:
@janzieg hi, if you load the 3s configuration and set to ulp mode, the sleep function will be changed to 3sec automatically, then if there is a ulp+ trigger, after 3s the ulp+ results will comes out. if there is no ulp+ trigger, the next data will come out after 300sec. so loading the 3s configuration means the bsec framework will check the task every 3sec, if set to LP, the measurement will done every 3sec, if set to ULP, the measurement will be done every 300seconds. the configuration and LP/ULP are 2 different parameters. this is quite confusing with the LP and ULP Trigger periods.