boschsensortec/BSEC-Arduino-library #6: BME680 and ThingSpeak

oleksid, 2018-04-24:
Hello.

I am sending the BME680 information to ThingSpeak but the IAQ Accuracy is always 0 and don't update.

Why is it happening?

Thanks,
Oleksandra S.

BoschSensortec, 2018-04-24:
Hi @oleksid,

It could be because you are violating the measurement time slots. Can you check the execution duration of the Thingspeak callback?

oleksid, 2018-04-24:
This is part of the code:
  client.print("POST /update HTTP/1.1\n");
        client.print("Host: api.thingspeak.com\n");
        client.print("Connection: close\n");
        client.print("X-THINGSPEAKAPIKEY: "+apiKey+"\n");
        client.print("Content-Type: application/x-www-form-urlencoded\n");
        client.print("Content-Length: ");
        client.print(postStr.length());
        client.print("\n\n");
        client.print(postStr);    
    }
    client.stop(); 
    delay(60000);

Is the execution duration of the Thingspeak callback 60 seconds?

oleksid, 2018-05-21:
Hello.

Do you think that if I use the ULP power mode, the IAQ values will update and send to ThingSpeak?

Thanks.