boschsensortec/BSEC-Arduino-library #7: IntegerDivideByZero in non-arduino esp32 (standard esp32-idf) when calling 'bsec_sensor_control'

pushtoopen, 2018-05-03:
I am NOT using the arduino firmware or backedn. I'm attempting to integrate bme680 libraries and the bsec binary into esp32. I've gotten fairly far but i believe i've run  into a sticking point by now.

Is the binary provided with bsec compiled using standard esp32-idf 3.0?

im using the included binary for esp32 from the bsec software download. i'm using the latest esp32-idf firmware.

i get the following error out when beginning to attempt to run bsec_iot_loop. i checked to ensure i wasn't feeding the function contained in the bsec binary a 0 value that may ber causing it and i nthe printf's above the error, i see no issues. 
I2C Installed (ran standard esp32 i2c init)
BME680 about to init (printf before "bsec_iot_init' function run)
BME680 Start init (populated the standard function struct 'bme680_g')
BME680_init start (ran 'bme680_init')
inside bme680_init so start null ptr check (ran 'null_ptr_check' had to remove comparing the struct itself to a NULL (if dev = NULL), this caused crashing)
passed null ptr check in bme_280init it returned 0
bme680_init finishedbsec_sensor_control inputs, time_stamp:26000, sensor_settings address: 1073490272 (LAST PRINTF BEFORE bssec_sensor_control FUNCTION IS RUN)

Guru Meditation Error: Core  1 panic'ed (IntegerDivideByZero)
. Exception was unhandled.
Core 1 register dump:
PC      : 0x400e41d0  PS      : 0x00060530  A0      : 0x800e3cbc  A1      : 0x3f
fce280
0x400e41d0: bsec_Bsec_checkIfSensorRequired at ??:?

A2      : 0x3ffc2f80  A3      : 0x00000001  A4      : 0x3ffc2960  A5      : 0x3f
fce2a2
A6      : 0x3ffce2a0  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x00
000000
A10     : 0x3ffcbc00  A11     : 0x00000004  A12     : 0x3ffce1f4  A13     : 0x3ffc324c
A14     : 0x00000000  A15     : 0x00000001  SAR     : 0x00000004  EXCCAUSE: 0x00000006
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff

Backtrace: 0x400e41d0:0x3ffce280 0x400e3cb9:0x3ffce2a0 0x400e3751:0x3ffce2f0 0x400d2a57:0x3ffce320 0x400d269c:0x3ffce340
0x400e41d0: bsec_Bsec_checkIfSensorRequired at ??:?

0x400e3cb9: bsec_codegen_sensor_control at ??:?

0x400e3751: bsec_sensor_control at ??:?

0x400d2a57: bsec_iot_loop at C:/msys32/home/esp/WORK/ESP32_Base/main/bsec_integration.c:456

0x400d269c: bme680sensor_task at C:/msys32/home/esp/WORK/ESP32_Base/main/bme680_tasks.c:261

CPU halted.

BoschSensortec, 2018-05-03:
Hi @pushtoopen,

It is hard to understand the problem from the description here. 

What were your original source files and what have you modified to fit your platform?

pushtoopen, 2018-05-04:
original files: bme680.c, bme680.h, bme680_defs.h, bsec_datatypes.h,bsec_interface.h,libalgobsec.a,libalgobsec.a.Size.log. libalgobsec.a is the esp32 version. My integration code uses the 'bsec_sensor_control' and feeds it with a bsec_bme_settings_t struct and the esp32 tick count ((void*)xTaskGetTickCount()*portTICK_PERIOD_MS). 

After the   'bsec_sensor_control' function is run, the platform crashes and outputs the above, specifically the items of interest are: '(IntegerDivideByZero)' and 'bsec_Bsec_checkIfSensorRequired at ??:?'

i dont have access to bsec_Bsec_checkIfSensorRequired function as it exists in the bsec pre-complid binarys. I suspect that the binary is not compiled using the esp-idf 3.0 platform.

BoschSensortec, 2018-05-04:
Hi @pushtoopen,

Are you using the `xtensa-esp32-elf-gcc` compiler?

pushtoopen, 2018-05-04:
Yes

pushtoopen, 2018-05-10:
[ESP32_BME680_Integration.zip](https://github.com/BoschSensortec/BSEC-Arduino-library/files/1990269/ESP32_BME680_Integration.zip)

Hi there, attached is cleaned up code for bme_680 integration with the esp32, esp_idf 3.0 (https://github.com/espressif/esp-idf/tree/release/v3.0) It compiles (windows 10 environment using msys32)) & still throws the same dividebyzero error

Darcy123, 2018-05-14:
Hi @pushtoopen,

I am the software engineer from Bosch Sensortec responsible for environmental sensor.
The lib on the website is indeed compiled by xtensa-esp32-elf-gcc.
Can you please write here your compiling option in your project?
Thanks!

pushtoopen, 2018-05-14:
Optimization Level: "Debug (-0g)"
Assertion Level: (Enabled)
Enable C++ exceptions: (unselected)
Stack smashing protection mode: (None)

please note that my compilation options and sdkconfig file are observable in the attached .zip file with all of my integration code listed above. You should be able to take my zip file above and build the project, flash it to a 'WROOM' esp32 chip & be able to see the same results as me using 'make monitor'