boschsensortec/BSEC-Arduino-library #9: warning code : 10

yukimach, 2018-05-05:
Is there documentation on what the warning code represents? 

I can't locate it to figure out what it means.

BoschSensortec, 2018-05-07:
@yukimach,

What warning code do you get??

BoschSensortec, 2018-05-08:
Hi @yukimach,

BSEC error codes can be found [here](https://github.com/BoschSensortec/BSEC-Arduino-library/blob/0255bd177335e9ac3557c30193b98ceac4cd564b/src/inc/bsec_datatypes.h#L282) and BME680 error codes can be found [here](https://github.com/BoschSensortec/BSEC-Arduino-library/blob/0255bd177335e9ac3557c30193b98ceac4cd564b/src/bme680/bme680_defs.h#L131).

The error you are receiving is a BSEC warning code

>  BSEC_W_SU_UNKNOWNOUTPUTGATE = 10, /*!< Output (virtual) sensor id passed to bsec_update_subscription() is not in the valid range; e.g., n_requested_virtual_sensors > actual number of output (virtual) sensors requested */

Can you describe your call to updateSubscription?

BoschSensortec, 2018-05-09:
Hi @yukimach,

Can you try with the latest version of the library 1.4.6.0?

yukimach, 2018-05-10:
Ok, looking into it. I've got the latest commit. library.properties file indicate 1.4.6.0 and when I check on arduino manage libraries it is also 1.4.6.0. Not sure why so I'm digging into it but if you have any thoughts on why this might be please let me know.

yukimach, 2018-05-10:
I had the previous libalgobsec. followed the new install set up and its all good now with no warning.
Thanks.

closed 2018-05-10 by yukimach