boschsensortec/BSEC-Arduino-library #11: Step 4. Modify the platform.txt file //missing file location

lipoforall, 2018-05-16:
Please add location of platform.txt file to manual.. I have spent about 4 hours finding out where the problem is.. I modified platform.txt file located in "C:\Program Files (x86)\Arduino\hardware\arduino\avr" instead the correct one located in:

for Arduino Mega: "C:\Users\username\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.6.21"

for ESP8266 "C:\Users\username\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1"

also there is working exaple of line "68-Combine gc-sections, archives, and objects" for MEGA 
`recipe.c.combine.pattern="{compiler.path}{compiler.c.elf.cmd}" {compiler.c.elf.flags} -mmcu={build.mcu} -o "{build.path}/{build.project_name}.elf" {object_files} {compiler.c.elf.extra_flags} "{build.path}/{archive_file}" "-L{build.path}" -lm`

anyway thanks for nice and very helpful library ;)

per1234, 2018-05-17:
As you discovered, it can be in a different location depending on which hardware package, how it was installed, and which OS you're using.

For example, if you hadn't installed Arduino AVR Boards 1.6.21 via Boards Manager then C:\Program Files (x86)\Arduino\hardware\arduino\avr\platform.txt would have been correct.

"Arduino15" folder locations on different OS:
- macOS: `/Users/{username}/Library/Arduino15`
- Windows:
  - Arduino IDE 1.6.5r5 and previous: `C:\Users\{username}\AppData\Roaming\Arduino15`
  - Arduino IDE 1.6.6 and later: `C:\Users\{username}\AppData\Local\Arduino15`
  - Using Windows App Store version of the IDE: `C:\Users\{username}\Documents\ArduinoData`
- Linux: `/home/{username}/.arduino15` (AKA `~/.arduino15`)

Manually installed hardware packages (such as ESP32) will be located under `{sketchbook folder/hardware` where `{sketchbook folder}` is set in the Arduino IDE at **File > Preferences > Sketchbook location**.

BoschSensortec, 2018-05-17:
@lipoforall,

In addition, you can enable 'Show verbose output during compilation' and you can see the paths of the core files compiled.

ferazambuja, 2018-05-21:
I had the same issue. The documentation should do a better job clarifying it.
It's way easier on Atom + Plataformio IDE, you just add `build_flags = -L/Users/USER/Documents/PlatformIO/Projects/BSEC_BME680/lib/ -lalgobsec` at the end of your project's platformio.ini. (however you still need the changes for the 8266)

henfri, 2018-10-11:
Hello,

I agree: The documentation is very clear except for the location of the platform.txt and the eagle* file.
Please add documentation on where these files are, or where this is documented.

Regards,
Hendrik