boschsensortec/BSEC-Arduino-library #12: IAQ accuracy always 0 after restart

michapr, 2018-05-25:
Using the lib at ESP8266, it is working stable.

Have changed the 
https://github.com/BoschSensortec/BSEC-Arduino-library/blob/master/examples/basic_config_state/basic_config_state.ino#L140
```
  if (stateUpdateCounter == 0) {
    /* First state update when IAQ accuracy is >= 1 */
    if (iaqSensor.iaqAccuracy >= 3) {
      update = true;

```
to
```
 if (stateUpdateCounter == 0) {
    /* First state update when IAQ accuracy is >= 1 */
    if (iaqSensor.iaqAccuracy >= 1) {      // <== was set to 1
      update = true;
```

All was working fine a while - but after a restart of sensor the IAQ accuracy came not back from "0" - never...
```
BSEC library version 1.4.6.0
Reading state from EEPROM
0 6 4 1 3D 0 0 0 0 0 0 0 29 0 0 0 14 0 1 3F 40 3B F2 2C 70 BB 15 40 C0 5E 52 48 A3 A2 14 40 C 0 2 3F 27 43 9 43 83 D4 37 43 9 0 5 3F 0 0 0 0 1 0 0 0 0 6F 1F 0 0 
Timestamp [ms], raw temperature [⸮C], pressure [hPa], raw relative humidity [%], gas [Ohm], IAQ, IAQ accuracy, temperature [⸮C], relative humidity [%]
341, 25.88, 101766.01, 37.48, 41594.13, 104.39, 0, 23.38, 43.52
3339, 25.63, 101768.37, 37.39, 38992.07, 111.75, 0, 23.10, 44.04
....
148755934, 25.85, 101683.50, 39.05, 37520.68, 129.62, 0, 23.30, 45.55
148758934, 25.86, 101682.16, 39.03, 37622.08, 129.63, 0, 23.28, 45.55

```
You see - after 148758934ms (so about 41 hours) nothing  happen. I already made a reset before, because the same was 2 days ago: after 3 days  I had same accuracy of 0.

So 2 questions - why this can happen? And, what I should do? Erase the EEPROM and will loose all calibration data from last 7 days?

Thanks!

BoschSensortec, 2018-06-01:
Hi @michapr,

That is the expected behavior after loading the state. In the current version, the calibration is performed at the start but unlike a fresh start, you will be able to read the IAQ index.

michapr, 2018-06-01:
No, this is not right.
In much other cases the IAQ indes is "0" only for a short time - but never for 3 days or more (now already for 10 days!) In such a case you neverwill have a chance to get an index >11 more - and new state will never be saved more.

So with current version there is no chance to save more any new state.

michapr, 2018-06-03:
I have cleared EEPROM and began at new - now all is working again as before.
If state "1" was saved (only after correction of the line in sample code - see above) I can reboot ("loss power") , will get at first state "0"  and after some short time state "1" again.

@BoschSensortec  Not sure what kind of issue this was - if you like you can close it, I cannot report more details now.