boschsensortec/BSEC-Arduino-library #14: ULP / ULP Plus mode - loop time limit

michapr, 2018-06-03:
What is the max delay in a loop - using the sample code configuration for ULP / ULP Plus (or SLP) ?

If there are needs to transfer the values via WiFi to a gateway it can take some time.
So I get sometimes the warning code 100:

`BSEC_W_SC_CALL_TIMING_VIOLATION = 100, /*!< Difference between actual and defined sampling intervals of bsec_sensor_control() greater than allowed */
`

Can you tell us something about this limit, or how better realize it ?
Code is simple - only for reference:
```
void loop(void)
{
  if (iaqSensor.run()) { // If new data is available
    output = String(millis());
    output += ", " + String(iaqSensor.rawTemperature);
    output += ", " + String(iaqSensor.pressure);
    output += ", " + String(iaqSensor.rawHumidity);
    output += ", " + String(iaqSensor.gasResistance);
    output += ", " + String(iaqSensor.iaqEstimate);
    output += ", " + String(iaqSensor.iaqAccuracy);
    output += ", " + String(iaqSensor.temperature);
    output += ", " + String(iaqSensor.humidity);
    Serial.println(output);
    updateState();
  } else {
    checkIaqSensorStatus();
  }
  if(send_param==1){
    wifi_send(iaqSensor.iaqEstimate,iaqSensor.temperature,iaqSensor.humidity,iaqSensor.pressure);
  }
 }
```
Thanks!

sysizlayan, 2018-07-03:
I am taking the same warning code when I load the given configuration files to library. Are you doing something similar in your setup?

BoschSensortec, 2018-07-03:
Hi @michapr,

The maximum delay time is dependent on the power mode. It is 3 seconds for the LP mode and 3 minutes for the ULP mode.

BoschSensortec, 2018-07-03:
Hi @sysizlayan,

What warning code do you receive when you load a configuration? Make sure that the configuration string matches the release of the static library in use. A mismatch would throw a warning or error.

sysizlayan, 2018-07-03:
I am just using the example code the library provided. I just load the values given in generic_33v_3s_28d folder to code and use it just like given in the example.

Then I get continuous warning#100 messages. I will provide the code as soon as I have free time.