boschsensortec/BSEC-Arduino-library #15: Bsec Arduino with ESP32

krish2487, 2018-06-11:
Hello,

First of, a huge thank you for the amazing sensor and now this library!! :-)

I have a rather peculiar issue with the BSEC library and ESP32.
We are using this sensor on a custom board with the esp32 and the Arduino core on the I2c bus.
We have the SDA and SCL lines on IO19 and 21 respectively.
The address is the default address (0x76).
We have been using the adafruit BME680 library (which itself is using the bsec headers for communicating with the sensor).

We have been successful so far and have been able to get the T/P/H and gas resistance values.

We are wanting to acquire even the AQI values (which is the reason for choosing the sensor in the first place) and this repo seems to be ideal for that. However we are unable to make it work.

A little information about our toolchain and setup.
We are platformio for development (basically a nice frontend that handles the nitty gritty details of Arduino or many such frameworks on the backend for us).

As I understand it, to integrate the library into the compilation process and execute the example code, all I need to do is to link the library libalgobsec.a into the binary by providing the path to the .a file.

platformio allows us to specify the linker flags by specifying them in the "build_flags" section.
and this is what it looks like.
"build_flags = -Llib/bsec/src/esp32 -lalgobsec"
where the -L is the path to the libalgobsec.a file.
and the bsec folder itself is the renamed folder of this repo.
And I have further copied the correct libalgobsec.a file from the bsec zip file I have downloaded from Bosch website.

I know that the compilation and linking process is happening since the binary is generated.
commenting out the build_flags line gives this error

```
.pioenvs/hornbill32dev/lib1c2/libbsec.a(bsec.cpp.o):(.literal._ZN4Bsec11beginCommonEv+0x0): undefined reference to `bsec_init'
.pioenvs/hornbill32dev/lib1c2/libbsec.a(bsec.cpp.o):(.literal._ZN4Bsec11beginCommonEv+0x4): undefined reference to `bsec_get_version'
.pioenvs/hornbill32dev/lib1c2/libbsec.a(bsec.cpp.o):(.literal._ZN4Bsec18updateSubscriptionEP21bsec_virtual_sensor_thf+0x0): undefined reference to `bsec_update_subscription'
.pioenvs/hornbill32dev/lib1c2/libbsec.a(bsec.cpp.o):(.literal._ZN4Bsec15readProcessDataEx19bsec_bme_settings_t+0x8): undefined reference to `bsec_do_steps'
.pioenvs/hornbill32dev/lib1c2/libbsec.a(bsec.cpp.o):(.literal._ZN4Bsec3runEv+0x0): undefined reference to `bsec_sensor_control'
.pioenvs/hornbill32dev/lib1c2/libbsec.a(bsec.cpp.o): In function `Bsec::beginCommon()':
/Users/srikrishnachaitanyanarumanchi/Documents/PlatformIO/Projects/bsec_test/lib/bsec/src/bsec.cpp:356: undefined reference to `bsec_init'
.pioenvs/hornbill32dev/lib1c2/libbsec.a(bsec.cpp.o): In function `Bsec::getVersion()':
/Users/srikrishnachaitanyanarumanchi/Documents/PlatformIO/Projects/bsec_test/lib/bsec/src/bsec.cpp:356: undefined reference to `bsec_get_version'
.pioenvs/hornbill32dev/lib1c2/libbsec.a(bsec.cpp.o): In function `Bsec::updateSubscription(bsec_virtual_sensor_t*, unsigned char, float)':
/Users/srikrishnachaitanyanarumanchi/Documents/PlatformIO/Projects/bsec_test/lib/bsec/src/bsec.cpp:356: undefined reference to `bsec_update_subscription'
.pioenvs/hornbill32dev/lib1c2/libbsec.a(bsec.cpp.o): In function `Bsec::readProcessData(long long, bsec_bme_settings_t)':
/Users/srikrishnachaitanyanarumanchi/Documents/PlatformIO/Projects/bsec_test/lib/bsec/src/bsec.cpp:292: undefined reference to `bsec_do_steps'
.pioenvs/hornbill32dev/lib1c2/libbsec.a(bsec.cpp.o): In function `Bsec::run()':
/Users/srikrishnachaitanyanarumanchi/Documents/PlatformIO/Projects/bsec_test/lib/bsec/src/bsec.cpp:175: undefined reference to `bsec_sensor_control'
collect2: error: ld returned 1 exit status
```

So I know that the linking step is working just fine.

However upon flashing the binary, I seem to get an error (-2) from the checkIaqSensorStatus() function as soon as the code prints the BSEC library version....

I understand that the -2 error is BME680_E_COM_FAIL as defined in the bme680_defs.h file.
which seems right as I did not specify the pins of the I2C bus (the default pins are different)

So I modified the line 102 of bsec.cpp 
Bsec::wireObj->begin();

to 

Bsec::wireObj->begin(19,21);

hoping to fix the error.
However the same error still persists. Is there some other step I am missing or some parameter I have configured wrong?

Any help is much appreciated.
:-)

Thank you in advance!!

krish2487, 2018-06-12:
Bump!!

I tried going by the book with the Arduino IDE as given in the documentation.
I installed the esp32 core on the Arduino IDE, replaced the Arduino-builder, modified the platform.txt file for the compiler.c.elf.extra_flags as described,  installed the BSEC.zip Arduino library and tried compiling the basic.ino sketch.

At first it gave a compilation error that it couldn't find the library under ~/Documents/Arduino/libraries/bsec/esp32.
And indeed the folder did not exist. I created the folder and dropped in the appropriate libalgobsec.a file from the BSEC folder user ESP32 ( Normal version) not the lite version and tried compiling.
It compiled and uploaded just fine. But I still got the same error.

This is my serial monitor output

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:812
load:0x40078000,len:0
load:0x40078000,len:11572
entry 0x40078a5c

BSEC library version 0.0.0.0
BME680 error code : -2
```

I assumed that it might be because of the pin allocation and tried specifying the pins as described in the above post but I still get the same error.

At least with the platformio process earlier, I was able to get the BSEC library version to print at 1.4.6.0. but with the Arduino IDE I am not able to get even that output.

Please help!!

krish2487, 2018-06-12:
Ok then. I feel stupid now.. 
The sensor was, in fact configured, for the I2C address to be 0x77.
modifying the address in the main.cpp fixed the "BME680 error code : -2"

The sketch is not able to progress to the next step and print one more line before failing at "BSEC error code : -2".

This is my serial monitor output.

```
BSEC library version 1.4.6.0
Timestamp [ms], raw temperature [°C], pressure [hPa], raw relative humidity [%], gas [Ohm], IAQ, IAQ accuracy, temperature [°C], relative humidity [%]
BSEC error code : -2
```

So any help on what I am goofing up now???

BSEC error code -2 indicates that 
`BSEC_E_DOSTEPS_VALUELIMITS = -2,                /*!< Value of input (physical) sensor signal passed to bsec_do_steps() is not in the valid range */`

So any tips on what I am doing wrong now??

Thank you in advance again!! :-)

krish2487, 2018-06-12:
Okay,
Seems like Platformio was causing the issue.

Compiling with Arduino and uploading the basic sketch works just fine.
I will need to check what is the fundamental difference in platformio and Arduino.
Except for the Arduino-builder executable there is no other difference.

BoschSensortec, 2018-06-14:
Hi @krish2487,

The arduino-builder helps with the compilation of the code. Refer to this existing [thread](https://github.com/BoschSensortec/BSEC-Arduino-library/issues/11?_pjax=%23js-repo-pjax-container) on a fix for PlatformIO.

BoschSensortec, 2018-06-14:
@krish2487,

Do you have access to the whole build log? If so, can you add it here?

krish2487, 2018-06-14:
Ok. I really feel like an idiot now..
 Got it to run.
I had duplicates of bme680.c, bme680.h, bme680_defs.h even in the src folder.
I didn't pay any attention to them until I saw the tree output above. :-P
I forgot they are present even in the bsec library.

Removing the redundant files in the src folder under root and compiling it works.
Now I am able to get the sensor output as expected.

@BoschSensortec Thank you again for the sensor and this library.

closed 2018-06-14 by krish2487

cyberman54, 2018-12-26:
@krish2487 I still don't get BSEC integrated with platformio. Tried it with current version 1.4.7.1, but always get stuck with the "BSEC error -2".

Could you explain which files of Bosch BSEC .zip download you used and put where in the tree?

krish2487, 2018-12-26:
@cyberman54 
Hello,
Can you provide a little more information about the hardware and software you are using.
There are 2 types of error reported by the library
1. BSEC error codes
2. BME680 error codes.

As I mentioned earlier, a BSEC error code of -2 indicates a communication error. Maybe there are no pull up resistors?? or something else is pulling the bus down??
Or in the case of ESP32 you need to specify which pins you want to instantiate the I2C peripheral on.
the default pins are different from what we used on our board.

cyberman54, 2018-12-26:
@krish2487 it's BSEC error -2, not BME680 error -2.
So as far as i see this is not a communication error, but an BSEC input value error.
I figured out that this happenes in BSEC.cpp in line 291:

`status = bsec_do_steps(inputs, nInputs, _outputs, &nOutputs);
`

All other calls before this line are successful, return with BSEC_OK  or BME680_OK error status. So i assume i2c communications is okay.

I start i2c interface with

Wire.begin(GPIO_NUM_21, GPIO_NUM_22);
iaqSensor.begin(0x76, Wire);

I think this works, because if i change i2c addr to 0x77 the library throws a BME680 error at start.

krish2487, 2018-12-26:
Well, given the point to which you have managed to isolate the issue I can only suggest that the number of virtual sensors you have setup do not match with what you are trying to read.

This is the test code that we have tried. maybe it has something to do with the BSEC_virtual_sensor_t structure that you might need to adjust to your needs??

`#include "bsec.h"

// Helper functions declarations
void checkIaqSensorStatus(void);
void errLeds(void);

// Create an object of the class Bsec
Bsec iaqSensor;

String output;

// Entry point for the example
void setup(void)
{
  Serial.begin(115200);

  iaqSensor.begin(BME680_I2C_ADDR_SECONDARY, Wire);
  output = "\nBSEC library version " + String(iaqSensor.version.major) + "." + String(iaqSensor.version.minor) + "." + String(iaqSensor.version.major_bugfix) + "." + String(iaqSensor.version.minor_bugfix);
  Serial.println(output);
  checkIaqSensorStatus();

  bsec_virtual_sensor_t sensorList[7] = {
    BSEC_OUTPUT_RAW_TEMPERATURE,
    BSEC_OUTPUT_RAW_PRESSURE,
    BSEC_OUTPUT_RAW_HUMIDITY,
    BSEC_OUTPUT_RAW_GAS,
    BSEC_OUTPUT_IAQ_ESTIMATE,
    BSEC_OUTPUT_SENSOR_HEAT_COMPENSATED_TEMPERATURE,
    BSEC_OUTPUT_SENSOR_HEAT_COMPENSATED_HUMIDITY,
  };

  iaqSensor.updateSubscription(sensorList, 7, BSEC_SAMPLE_RATE_LP);
  checkIaqSensorStatus();

  // Print the header
  output = "Timestamp [ms], raw temperature [°C], pressure [hPa], raw relative humidity [%], gas [Ohm], IAQ, IAQ accuracy, temperature [°C], relative humidity [%]";
  Serial.println(output);
}

// Function that is looped forever
void loop(void)
{

  if (iaqSensor.run()) { // If new data is available
    Serial.println();
    Serial.println("rTemperature: " + String(iaqSensor.rawTemperature));
    Serial.println("Pressure: " + String(iaqSensor.pressure));
    Serial.println("rHumidity: " + String(iaqSensor.rawHumidity));
    Serial.println("Gas Resistance: " + String(iaqSensor.gasResistance));
    Serial.println("IAQ: " + String(iaqSensor.iaqEstimate));
    Serial.println("IAQ Accuracy: " + String(iaqSensor.iaqAccuracy));
    Serial.println("cTemperature: " + String(iaqSensor.temperature));
    Serial.println("cHumidity: " + String(iaqSensor.humidity));                
    Serial.print("ESP Free Heap: ");
    Serial.println(ESP.getFreeHeap()>>10);
//    output = String(millis());
//    output += ", " + String(iaqSensor.rawTemperature);
//    output += ", " + String(iaqSensor.pressure);
//    output += ", " + String(iaqSensor.rawHumidity);
//    output += ", " + String(iaqSensor.gasResistance);
//    output += ", " + String(iaqSensor.iaqEstimate);
//    output += ", " + String(iaqSensor.iaqAccuracy);
//    output += ", " + String(iaqSensor.temperature);
//    output += ", " + String(iaqSensor.humidity);
//    Serial.println(output);
  } else {
    checkIaqSensorStatus();
  }
}

// Helper function definitions
void checkIaqSensorStatus(void)
{
  if (iaqSensor.status != BSEC_OK) {
    if (iaqSensor.status < BSEC_OK) {
      output = "BSEC error code : " + String(iaqSensor.status);
      Serial.println(output);
      for (;;)
        errLeds(); /* Halt in case of failure */
    } else {
      output = "BSEC warning code : " + String(iaqSensor.status);
      Serial.println(output);
    }
  }

  if (iaqSensor.bme680Status != BME680_OK) {
    if (iaqSensor.bme680Status < BME680_OK) {
      output = "BME680 error code : " + String(iaqSensor.bme680Status);
      Serial.println(output);
      for (;;)
        errLeds(); /* Halt in case of failure */
    } else {
      output = "BME680 warning code : " + String(iaqSensor.bme680Status);
      Serial.println(output);
    }
  }
}

void errLeds(void)
{
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, HIGH);
  delay(100);
  digitalWrite(LED_BUILTIN, LOW);
  delay(100);
}`

cyberman54, 2018-12-27:
This is weird...
Which libalgobsec version fits to which bme680.c version and which bsec_datatypes.h / bsec_interface.h?

@BoschSensortec could you please update the documentation?

cyberman54, 2018-12-27:
I finally found the root cause and fixed it, see my PR #39 
See my PR #39 to fix this
https://github.com/BoschSensortec/BSEC-Arduino-library/pull/39

@krish2487 thanks for your hints, pointing me in right direction

@BoschSensortec obviously you did not test / update this code with later BSEC versions. I would have expected a better software quality management by Bosch. Took me several hours to get this fixed.

krish2487, 2018-12-27:
@cyberman54  glad to see you fix it! :-)

I did not do anything.. I was as much in the dark as you were. You did all the heavy lifting and fixes.
kudos!