boschsensortec/BSEC-Arduino-library #16: Bsec causes buffer overflow in ESP32

sysizlayan, 2018-06-21:
Hi,
I am currently working on a project using BME680 air quality sensor using ESP32 Arduino library.

In setup function, I initialize the sensor using SPI, updated subscriptions, restore the previous values. I am running the sensor in LP mode. I call "run" function in every loop iteration. If it returns true, I read the values then keep going. In short, I am doing exactly the same as example.
My problem is my code works fine for around 2-3 hours, but after that I am taking Stack smashing protect failure from ESP32 and system reboots itself. I first thought it is because of some buffer I am using, I disabled some of them one by one and tested for 2 days. It occured again.
The moment I get suspicious about the sensor is where I got continious reboots. During initialization whenever I restored the configuration, the system runs for several loops until the sensor data is ready and after sensor reading, I got stack smahing protect failure again.

My setup function:
`void setEnvironmentalSensorUp()
{
  bsec_virtual_sensor_t sensorList[5] = // Values to be subscribed
  {
    BSEC_OUTPUT_RAW_PRESSURE,
    BSEC_OUTPUT_RAW_GAS,
    BSEC_OUTPUT_IAQ_ESTIMATE,
    BSEC_OUTPUT_SENSOR_HEAT_COMPENSATED_TEMPERATURE,
    BSEC_OUTPUT_SENSOR_HEAT_COMPENSATED_HUMIDITY,
  };
  //Wire.begin(SDA, SCL, 400000); // 21 and 22 pins, 400Khz Freq
  //bme680.begin(BME680_I2C_ADDR_SECONDARY, Wire);
  SPI.begin(23, 21, 22, 5); // MISO: 21, MOSI: 22, SCK: 23, SS: 5, order sck,miso,mosi,ss
  bme680.begin(5, SPI);

  String fwVersion = "BSEC v" + String(bme680.version.major) + "." + String(bme680.version.minor) + "." + String(bme680.version.major_bugfix) + "." + String(bme680.version.minor_bugfix);
  DEBUG_UART.println(fwVersion);
  bme680.updateSubscription(sensorList, 5, BSEC_SAMPLE_RATE_LP); // 3 seconds refresh rate
  checkIaqSensorStatus(false);
  
 /*bme680.setConfig(bsec_config_iaq);
  checkIaqSensorStatus(false);*/
  
  restoreBME680Data(); // restore BME680 data from bme680.conf file
  nextStorageTimer = millis() + STATE_SAVE_PERIOD; // store the time when the status will be saved
}`

My restore function:
`void restoreBME680Data()
{
  File bme680Config;

  if (SPIFFS.exists("/BME680.conf")) // if the config file exists
  {
    bme680Config = SPIFFS.open("/BME680.conf", "r"); // Open the file
    if (bme680Config.size() == BSEC_MAX_STATE_BLOB_SIZE) // size check
    {
      bme680Config.read(bsecState, BSEC_MAX_STATE_BLOB_SIZE); //Read the file into bsecState
      bme680Config.close(); // close te file for future usage
      bme680.setState(bsecState); //take bme680 status into bsecState array
      checkIaqSensorStatus(true); // check if everything is ok

      log_i("BME680 configuration is restored!\r\n");
    }
    else
    {
      log_d("[BME680_RESTORE] BME680 config file size is not correct!");
    }
  }
  else
  {
    log_i("[BME680_RESTORE] BME680 config file does not exists!");
  }
}`

My storage function:
`bool storeBME680Data()
{
  File bme680Config;
  if (bme680.iaqAccuracy >= 1) // if the config file exists and sensor is in calibration mode
  {
    bme680.getState(bsecState); //take bme680 status into bsecState array

    checkIaqSensorStatus(false); // check if everything is ok
    bme680Config = SPIFFS.open("/BME680.conf", "w"); // Open the file
    bme680Config.write(bsecState, BSEC_MAX_STATE_BLOB_SIZE); //Write the array into BME680.conf file
    bme680Config.close(); // close te file for future usage

    log_i("BME680 config file is saved!");
    return true;
  }
  return false;
}`

My main loop handler function:
`void environmentalSensorHandler()
{
  if (bme680.run()) // If new data is available
  {
    log_i("New values\r\n\tIAQ Estimate(Accuracy): %.2f(%d)\r\n\tHumidity: %.2f\r\n\tTemperature: %.2fC\r\n\tPressure: %.2f\r\n\tGas resistance: %.2f\r\n",
          bme680.iaqEstimate,
          bme680.iaqAccuracy,
          bme680.humidity,
          bme680.temperature,
          bme680.pressure,
          bme680.gasResistance);

    enviromentalSensor.airQuality           = bme680.iaqEstimate;
    enviromentalSensor.airQuality_presicion = bme680.iaqAccuracy;
    enviromentalSensor.humidity             = bme680.humidity;
    enviromentalSensor.temperature          = bme680.temperature;
    enviromentalSensor.sensorGasResistance  = bme680.gasResistance;
    enviromentalSensor.pressure             = bme680.pressure;

    wasTheStateChanged = true; // set the status publish flag
  }
  else
  {
    checkIaqSensorStatus(false);
    if (millis() >= nextStorageTimer)
    {
      if (storeBME680Data())
        nextStorageTimer = millis() + STATE_SAVE_PERIOD; // store the next storage time
    }
  }
}`

My check sensor function:
`void checkIaqSensorStatus(bool deleteConfig)
{
  String output;
  if (bme680.status != BSEC_OK)
  {
    if (bme680.status < BSEC_OK)
    {
      output = "ERROR [BME_680] BSEC error code : " + String(bme680.status);
      DEBUG_UART.println(output);

      if (deleteConfig)
      {
        if (SPIFFS.remove("/BME680.conf"))
          log_i("Configuration deleted");
        else
          log_i("Delete failed");
      }
      delay(1000);
      ESP.restart();
    }
    else
    {
      output = "INFO [BME_680] BSEC warning code : " + String(bme680.status);
      DEBUG_UART.println(output);
    }
  }

  if (bme680.bme680Status != BME680_OK)
  {
    if (bme680.bme680Status < BME680_OK)
    {
      output = "ERROR [BME_680] BME680 error code : " + String(bme680.bme680Status);
      DEBUG_UART.println(output);
      if (deleteConfig)
      {
        if (SPIFFS.remove("/BME680.conf"))
          log_i("Configuration deleted");
        else
          log_i("Delete failed");
      }
      delay(1000);

      ESP.restart();
    }
    else
    {
      output = "INFO [BME_680] BME680 warning code : " + String(bme680.bme680Status);
      DEBUG_UART.println(output);
    }
  }
}`

My error trace:
`Stack smashing protect failure!

abort() was called at PC 0x400ef580 on core 1

Backtrace: 0x4008d9ec:0x3ffd9880 0x4008dbc7:0x3ffd98a0 0x400ef580:0x3ffd98c0 0x400d6c07:0x3ffd98e0 0x40150964:0x3ffd9920

Rebooting...`

The issue from esp32 arduino core library:
[https://github.com/espressif/arduino-esp32/issues/1455](url)

When I disabled the sensor and tested for 2 days, I got no stack smahing error.
My code is above, please check it if there is some problematic part in any of them. If not, please check the library.

Thanks in advance.

sysizlayan, 2018-07-03:
Is there any information about the issue?

BoschSensortec, 2018-07-03:
Hi @sysizlayan,

Do you have a proposal on how we can duplicate this issue on our side? Do you have a complete Arduino Sketch and instructions on how to setup the configuration file in the SPIFFS?

sysizlayan, 2018-07-03:
I will change my main code for you to test.

At the same time, could you please check if there exists a buffer which may cause buffer overflow?

rado99, 2018-12-04:
Hello,
How did you succesfully got this library for esp32? I ran it for esp8266 only, what I am doing wrong?