boschsensortec/BSEC-Arduino-library #18: IAQ and deepsleep on ESP32

Creepwood, 2018-07-11:
Hey there,
I am struggling with IAQ calculation on an ESP32 using deepsleep mode.
My plan is, to wakeup the ESP32 every 5 minutes, read the sensor data and go back to deepsleep.

I'm using the loadState() function after wakeup and call the updateState() function before going to deep sleep. But the IAQ-index and IAQ-accuracy stays always 0. Running it for weeks... but no changes.

Is there a possibility to realize plan?

BoschSensortec, 2018-07-12:
Hi @Creepwood,

Loading a state effectively restarts the calibration process. You hence observe the IAQ index at 0. The use-case mentioned by you will be supported in the upcoming release.

torntrousers, 2018-07-19:
Any timeframe for when a release supporting this will be available?

Creepwood, 2018-10-07:
Hi @BoschSensortec ,

should the feature still be developed?

BoschSensortec, 2018-10-08:
Hi @Creepwood,

The core library has already been released on our website. This repository will be updated soon.