boschsensortec/BSEC-Arduino-library #20: xtensa-esp32-elf/bin/ld.exe: cannot find -lalgobsec

jttaylor86, 2018-07-17:
arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot find -lalgobsec

collect2.exe: error: ld returned 1 exit status

Why does this occur?

closed 2018-07-17 by jttaylor86

torntrousers, 2018-07-19:
Hi @jttaylor86 I'm getting the same error about 'cannot find -lalgobsec' - you've closed this so how did you fix this?

jttaylor86, 2018-07-19:
Hi,

so I moved the extra compiler flags to the end of the line in the platform file.

Delete your current lines 76-77 and paste in this. 

## Combine gc-sections, archives, and objects
recipe.c.combine.pattern="{compiler.path}{compiler.c.elf.cmd}" {compiler.c.elf.flags} -Wl,--start-group {object_files} "{build.path}/arduino.ar" {compiler.c.elf.libs} {compiler.c.elf.extra_flags} -Wl,--end-group -Wl,-EL -o "{build.path}/{build.project_name}.elf"