boschsensortec/BSEC-Arduino-library #21: Teensy 3.2

BenDrewry, 2018-07-17:
Has anyone had success running the 'basic' program using the Teensy 3.2 microcontroller running MK20DX256?  When trying to run the code I get the following error message: 
Arduino: 1.8.5 (Windows 7), TD: 1.42, Board: "Teensy 3.2 / 3.1, Serial, 96 MHz (overclock), Faster, US English"

WARNING: library BSEC-Arduino-library-master claims to run on (samd, sam, esp8266, nrf52, esp32) architecture(s) and may be incompatible with your current board which runs on (avr) architecture(s).
GetFileAttributesEx C:\Program Files (x86)\Arduino\libraries\BSEC-Arduino-library-master\src\mk20dx256: The system cannot find the file specified.
Error compiling for board Teensy 3.2 / 3.1.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Thanks,
Ben

BenDrewry, 2018-07-17:
There is a Teensy extension that installs the hardware compiler under the avr folder.  Since the path is coming up as mk20dx256, since the Teensy is really an ARM, not AVR.  I renamed the source folder as mk20... and copied the contents of C:\...\BSEC_1.4.6.0_Generic_Release_20180425.zip\BSEC_1.4.6.0_Generic_Release_20180425\algo\bin\Normal_version\armcc\Cortex_M4 here.  Now I am getting the following errors:

WARNING: library BSEC-Arduino-library-master claims to run on (samd, sam, esp8266, nrf52, esp32) architecture(s) and may be incompatible with your current board which runs on (avr) architecture(s).
C:\Users\drewrbj1\AppData\Local\Temp\arduino_build_304261\libraries\BSEC-Arduino-library-master\mk20dx256\bsec.cpp.o: In function `Bsec::delay_ms(unsigned long)':

C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Wire/WireKinetis.h:135: multiple definition of `Bsec::delay_ms(unsigned long)'

C:\Users\drewrbj1\AppData\Local\Temp\arduino_build_304261\libraries\BSEC-Arduino-library-master\bsec.cpp.o:C:\Program Files (x86)\Arduino\libraries\BSEC-Arduino-library-master\src/bsec.cpp:396: first defined here

c:/program files (x86)/arduino/hardware/tools/arm/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/bin/ld.exe: Disabling relaxation: it will not work with multiple definitions

C:\Users\drewrbj1\AppData\Local\Temp\arduino_build_304261\libraries\BSEC-Arduino-library-master\mk20dx256\bsec.cpp.o: In function `Bsec::delay_ms(unsigned long)':

C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Wire/WireKinetis.h:135: multiple definition of `Bsec::i2cWrite(unsigned char, unsigned char, unsigned char*, unsigned short)'

C:\Users\drewrbj1\AppData\Local\Temp\arduino_build_304261\libraries\BSEC-Arduino-library-master\bsec.cpp.o:C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Wire/WireKinetis.h:132: first defined here

C:\Users\drewrbj1\AppData\Local\Temp\arduino_build_304261\libraries\BSEC-Arduino-library-master\mk20dx256\bsec.cpp.o: In function `Bsec::delay_ms(unsigned long)':

C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Wire/WireKinetis.h:135: multiple definition of `Bsec::i2cRead(unsigned char, unsigned char, unsigned char*, unsigned short)'

C:\Users\drewrbj1\AppData\Local\Temp\arduino_build_304261\libraries\BSEC-Arduino-library-master\bsec.cpp.o:C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Wire/WireKinetis.h:132: first defined here

C:\Users\drewrbj1\AppData\Local\Temp\arduino_build_304261\libraries\BSEC-Arduino-library-master\mk20dx256\bsec.cpp.o: In function `Bsec::delay_ms(unsigned long)':

C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Wire/WireKinetis.h:135: multiple definition of `Bsec::spiTransfer(unsigned char, unsigned char, unsigned char*, unsigned short)'

C:\Users\drewrbj1\AppData\Local\Temp\arduino_build_304261\libraries\BSEC-Arduino-library-master\bsec.cpp.o:C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Wire/WireKinetis.h:132: first defined here

C:\Users\drewrbj1\AppData\Local\Temp\arduino_build_304261\libraries\BSEC-Arduino-library-master\mk20dx256\bsec.cpp.o: In function `Bsec::delay_ms(unsigned long)':

C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Wire/WireKinetis.h:135: multiple definition of `Bsec::Bsec()'

C:\Users\drewrbj1\AppData\Local\Temp\arduino_build_304261\libraries\BSEC-Arduino-library-master\bsec.cpp.o:C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Wire/WireKinetis.h:132: first defined here

C:\Users\drewrbj1\AppData\Local\Temp\arduino_build_304261\libraries\BSEC-Arduino-library-master\mk20dx256\bsec.cpp.o: In function `Bsec::delay_ms(unsigned long)':

C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Wire/WireKinetis.h:135: multiple definition of `Bsec::Bsec()'

C:\Users\drewrbj1\AppData\Local\Temp\arduino_build_304261\libraries\BSEC-Arduino-library-master\bsec.cpp.o:C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Wire/WireKinetis.h:132: first defined here

C:\Users\drewrbj1\AppData\Local\Temp\arduino_build_304261\libraries\BSEC-Arduino-library-master\mk20dx256\bsec.cpp.o: In function `Bsec::delay_ms(unsigned long)':

C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Wire/WireKinetis.h:135: multiple definition of `Bsec::beginCommon()'

C:\Users\drewrbj1\AppData\Local\Temp\arduino_build_304261\libraries\BSEC-Arduino-library-master\bsec.cpp.o:C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Wire/WireKinetis.h:132: first defined here

C:\Users\drewrbj1\AppData\Local\Temp\arduino_build_304261\libraries\BSEC-Arduino-library-master\mk20dx256\bsec.cpp.o: In function `Bsec::delay_ms(unsigned long)':

C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Wire/WireKinetis.h:135: multiple definition of `Bsec::begin(unsigned char, bme680_intf, signed char (*)(unsigned char, unsigned char, unsigned char*, unsigned short), signed char (*)(unsigned char, unsigned char, unsigned char*, unsigned short), void (*)(unsigned long))'

C:\Users\drewrbj1\AppData\Local\Temp\arduino_build_304261\libraries\BSEC-Arduino-library-master\bsec.cpp.o:C:\Program Files (x86)\Arduino\hardware\teensy\avr\libraries\Wire/WireKinetis.h:132: first defined here

C:\Users\drewrbj1\AppData\Local\Temp\arduino_build_304261\libraries\BSEC-Arduino-library-master\mk20dx256\bsec.cpp.o: In function `Bsec::delay_ms(unsigned long)':

[... cut ...]
C:\Users\drewrbj1\AppData\Local\Temp\arduino_build_304261\libraries\BSEC-Arduino-library-master\bsec.cpp.o: In function `Bsec::readProcessData(long long, bsec_bme_settings_t)':

C:\Program Files (x86)\Arduino\libraries\BSEC-Arduino-library-master\src/bsec.cpp:291: undefined reference to `bsec_do_steps'

C:\Users\drewrbj1\AppData\Local\Temp\arduino_build_304261\libraries\BSEC-Arduino-library-master\bsec.cpp.o: In function `Bsec::run()':

C:\Program Files (x86)\Arduino\libraries\BSEC-Arduino-library-master\src/bsec.cpp:174: undefined reference to `bsec_sensor_control'

collect2.exe: error: ld returned 1 exit status

Error compiling for board Teensy 3.2 / 3.1.

Hiawathaa, 2018-08-22:
Hi,
I have the same problem and I think it might be because I haven't edited the platform.txt at 
C:\Program Files (x86)\Arduino\hardware\teensy\avr. What should it be for teensy? Currently it is like this:

## Link
#recipe.c.combine.pattern="{compiler.path}{build.toolchain}{build.command.gcc}" {build.flags.optimize} {build.flags.ld} {build.flags.ldspecs} {build.flags.cpu} -o "{build.path}/{build.project_name}.elf" {object_files}"{build.path}/{archive_file}"  "-L{build.path}" {build.flags.libs}

There is also a platform.txt at 
C:\Program Files (x86)\Arduino\hardware\arduino\avr but changing it made no difference?

## Combine gc-sections, archives, and objects
recipe.c.combine.pattern="{compiler.path}{compiler.c.elf.cmd}"  -mmcu={build.mcu} -o "{build.path}/{build.project_name}.elf" {object_files} "{build.path}/{archive_file}" {compiler.c.elf.extra_flags} "-L{build.path}" -lm