boschsensortec/BSEC-Arduino-library #27: BSEC error code -15 on Arduino Zero

Adminius, 2018-10-08:
Hi,

thanks for installation instruction.

i've tried BME680 with Adruino Zero (Core 1.6.19) and Arduino IDE 1.8.5 the basic example.
I get this error (LED blinks):
```
BSEC library version 1.4.7.1
BSEC error code : -15
```
if i disable this  [checkIaqSensorStatus() call](https://github.com/BoschSensortec/BSEC-Arduino-library/blob/master/examples/basic/basic.ino#L33)
i get a normal values:
```
429900, 25.00, 98555.87, 52.34, 194579.12, 30.12, 1, 24.94, 52.54
432900, 25.00, 98557.47, 52.44, 194579.12, 31.87, 1, 24.94, 52.62
435900, 25.01, 98557.02, 52.56, 193891.80, 34.26, 1, 24.94, 52.74
438900, 25.00, 98556.62, 52.74, 193777.72, 35.98, 1, 24.94, 52.94

```

i've used *libalgobsec.a* for m0+

what's wrong with it?

i've also tried to change sensorList array size and **7** to another value... allways the same error code

Skel0t, 2018-10-14:
Hi, I've tried BME680 with ESP32 (Heltech WIFI Kit 32) and Arduino IDE 1.8.5 and the basic example. 
I am getting the same error as you do.

I fixed the issue by adding 1 to nSensorSettings and to the size of SensorSettings in the method `updateSubscription(bsec_virtual_sensor_t sensorList[], uint8_t nSensors, float sampleRate);` since error code -15 says it had to be bigger. After that it worked. I just don't know yet if that breaks anything else. But I could not find any other way yet to solve this.

ITGuyDave, 2018-10-30:
@BoschSensortec has traditionally been fairly active in the Issues tracker - but have been conspicuously very quiet on this particular issue. I have yet to find a solution to this (other than the above), although cannot understand why this issue has suddenly appeared in the first case. I am using a MKR1000 (essentially a Arduino Zero), and have followed the instructions word for word with zero success. I've never been more frustrated coding for Arduino!

I've tried following every possible tutorial to the letter, but have not had any luck whatsoever with the BSEC library and the BME680. I've virtually given up on the sensor, as the IAQ functionality I purchased it for is only available with the BSEC library that I cannot get to work no matter what I do! Always the -15 error, BSEC_E_SU_GATECOUNTEXCEEDSARRAY when following the tutorial word for word.

Please @BoschSensortec, help us out as BSEC is a compiled library that is virtually impossible to troubleshoot ourselves!

BoschSensortec, 2018-10-30:
Hi @ITGuyDave,

The subscription of inputs is not in sync with the release of BSEC library. Note that only updating the static library is not sufficient. you will also need to update the header files as they are updated too between versions. BSEC_OUTPUT_IAQ_ESTIMATE is now renamed to BSEC_OUTPUT_IAQ with the same value and hence you see an output even though the header files are incorrect.

In addition, you may subscribe to new outputs, namely `BSEC_OUTPUT_STATIC_IAQ`, ` BSEC_OUTPUT_CO2_EQUIVALENT`, `BSEC_OUTPUT_BREATH_VOC_EQUIVALENT`, `BSEC_OUTPUT_COMPENSATED_GAS`, `BSEC_OUTPUT_GAS_PERCENTAGE`. Details of which are present in bsec_datatypes.h

ITGuyDave, 2018-10-31:
@BoschSensortec based on your response, it appears that somehow I have a de-synced combination of the compiled libraries and the headers. 

The process I had followed is:
1. Installed Arduino IDE 1.8.7
2. Downloaded BSEC from this GitHub repository
3. Replaced the Arduino builder with that mentioned in the instructions here (version 219)
4. Edited the platform.txt for the relevant platform
5. Copied libalgobsec.a to the cortex-m0plus subfolder in the library "src" folder, from the ZIP provided on the Bosch website as per instructions
6. Attempted to compile the "basic" example (which still refers to the BSEC_OUTPUT_IAQ_ESTIMATE).

Thanks to you pointing out the change in definitions, looking at the various bsec_datatypes.h files - I was able to finally determine that there is an omission/error in the BSEC instructions on this repo, which resulted in me using the wrong/outdated library. Essentially, replacing the headers in the "inc" folder, along with the other bsec.h and bsec.cpp files and the examples - I was able to get this working!

I'm not sure if I could simply use the BSEC library part of the ZIP from the Bosch site (in the Arduino folder), or if I should use the libraries from this repository. Could you please update the instructions to ensure this is properly communicated on installing the most current release? I would also be interested in knowing whether I should be separately maintaining the BME680 library from the GitHub Repo, or waiting for the releases of the BSEC library on the Bosch site. Could you please advise the release/update process?

Thanks for the help @BoschSensortec!

henfri, 2018-11-11:
Hello,

I have the same issue. I
I do not know how anything could have become out of sync. I installed everything on the same day.
Also:
@BoschSensortec : 

> BSEC_OUTPUT_IAQ_ESTIMATE is now renamed to BSEC_OUTPUT_IAQ

But here https://github.com/BoschSensortec/BSEC-Arduino-library/blob/0255bd177335e9ac3557c30193b98ceac4cd564b/examples/basic/basic.ino#L27

still BSEC_OUTPUT_IAQ_ESTIMATE  is used.

So, for me with a fresh install this example does not work.

Regards,
Hendrik

Oandr, 2019-02-04:
Hi,

I've been trying this since for quite some time on a NodeMCU8266 V3 and still have the erroneous message BSEC error code : -15. The fix from Skel0t is also not working. I've attached the BME680 at the secondary address 0x77. Tried it on 2 Arduino IDE 1.8.5 (Win7) and 1.8.8 (MacOS) - same result.

I realized during the evaluation that the proposed change to linker script eagle.app.v6.common.ld is invalidated every compilation since in the standard platform.txt there is 
```
## Build the app.ld linker file
recipe.hooks.linking.prelink.1.pattern="{compiler.path}{compiler.c.cmd}" -CC -E -P {build.vtable_flags} "{runtime.platform.path}/tools/sdk/ld/eagle.app.v6.common.ld.h" -o "{runtime.platform.path}/tools/sdk/ld/eagle.app.v6.common.ld"
```

So that I could avoid by commenting this recipe ...

Is there anything which I could try to resolve that issue?

Regards
Andreas

BoschSensortec, 2019-02-05:
Hi @Oandr,

There is currently a pull-request under review, the following is the proposed change to the linker script modifications.

> In previous versions of this manual it was stated that you need to edit `eagle.app.v6.common.ld` directly. Since this file is now generated at build time you will need to change the linker script file, `eagle.app.v6.common.ld.h` typically found in `{YourESP8266PPackageDirectory}\tools\sdk\ld`.