boschsensortec/BSEC-Arduino-library #31: Change i2c Pins

henfri, 2018-11-10:
Hello,

I need to change the I2C pins the Sensor is connected to, as the default Pins are used by other Hardware on my ESP-Board (Witty-Board).

I found this:
"Wire – The ESP8266 should work with any I2C sensor you can throw at it – just use the same Wire API calls you’re used to. There are a few differences:
Pin definition: The ESP2866 doesn’t actually have any hardware I2C pins – those labeled on the Thing are the default, but you can actually use any two pins as SDA and SCL. Calling Wire.begin() will assume pins 2 and 14 are SDA and SCL, but you can manually set them to any other pin by calling Wire.begin([SDA], [SCL])."

but I cannot find Wire.begin in the code. I assume the Library does not use the Wire library.

Is there another way to change the GPIO ports that the Sensor is connected to in the Software?

Regards,
Hendrik

febinjohnk, 2018-11-26:
Hederic 

1.)go to BSEC-Arduino-library
"\Documents\Arduino\libraries\BSEC-Arduino-library-master"
2.)find bsec.cpp
3.)line number102
change to 
Bsec::wireObj->begin(D1,D2);//

Regards 
Febin

henfri, 2018-11-30:
Hello,

Thanks.
I found now, that it also works without changing the code of the library.
Just use wire.begin(Pin1, Pin2)

Regards,
Hendrik