boschsensortec/BSEC-Arduino-library #32: Error code -34

henfri, 2018-11-11:
Hello,

when running basic_config_state, I get error code -34.

I found about this error:

> 3.3.3.1 BSEC_E_CONFIG_VERSIONMISMATCH
> Possiblereasons:
> · General description: BSEC_E_CONFIG_VERSIONMISMATCH
> · Version mentioned in the conﬁguration string or state string passed to bsec_set_conﬁguration() or bsec_setstate(),respectively, does not match with the currentversion. Correction methods:
> · Obtain a compatible string.
> · Acallto bsec_get_version() would give the current version information.

But sorry, I don't know what to do here. Could you elaborate please, what is going on?

Regards,
Hendrik

tedani, 2018-12-29:
I have the same problem.

basic_config_state.ino gives BSEC error code -34
basic.ino gives BSEC error code -15

It seems the development team is too busy to prepare for Christmas holiday.

cyberman54, 2018-12-29:
BSEC documentation says: "Please note that configuration and state strings are coded to a specific version and will not be accepted by other versions of bsec."
SInce current BSEC version 1.4.7.1 is newer than BSEC version this Arduino Library was compiled for, you probably get BSEC error -34 because the code tries to load serialized configuration or state for older BSEC version.

cyberman54, 2018-12-29:
You are right, the code is currently not usable as is.

cyberman54, 2018-12-29:
@henfri yes, better way is to use everything from the current BSEC-Zip. 
But bsec.pp from the Arduino Library will not work properly with current BSEC. I created [a pull request](https://github.com/BoschSensortec/BSEC-Arduino-library/pull/39) to fix this. As long as the PR is not merged, you can manually merge it to your local bsec.cpp.

tedani, 2018-12-29:
@cyberman54 Thanks.  I didn't read the doc (you know, engineers don't read manuals).  The information is in BSEC_1.4.7.1_Generic_Release_20180907/integration_guide/BST-BME680-Integration-Guide-AN008-47.pdf

henfri, 2018-12-29:
Hello,

Thanks, but I did fix it by using the code from the Bosch-Homepage.

Greetings,
Hendrik