boschsensortec/BSEC-Arduino-library #45: A lot of questions/Integration guide

Kamairo, 2019-01-20:
Hi all,

I find the Integration Guide somewhat confusing, so here are my questions:

- [ ] What*s the difference between setting a configuration via ``bsec_set_configuration()`` and setting a sample rate via ``bsec_update_subscription()``? I noticed that even when I load a 300s config, I can change the subscription to 3s and I will get a new value every 3s. See ``basic_config_state_ULP_LP`` for example.
- [ ] What does exactly change between the 3.3V and the 1.8V configs? I mean, I can't change the voltage by software?! So what happens if I supply 3.3V to the BME and set the config to 1.8V and vice versa?
- [ ] What happens if I do not let the library get a measurement in 1.5 x sample rate (e.g. 300s x 1.5 = 450s)? The IAQ accuracy just remains 0?
- [ ] Just checking if I understood correctly: I need to save the state of the library to some sort of non-volatile memory, because otherwise the state will reset on each boot?

I have my own PCB for a ESP8266 and I am using these controllers to collect sensor readings every 25 min and then they go to deep sleep, since they run on battery. What I've read so far in the integration guide and here on github, this is not possible with the BME680 and BSEC?! I would need to reduce the wake up periods to at least 450s - the time it takes for the ESP to get the reading to the library?! Plus I would need to save the state of BSEC to EEPROM before going into deep sleep and load the state after waking up each time, is that correct?

cyberman54, 2019-01-20:
> I need to save the state of the library to some sort of non-volatile memory, because otherwise the state will reset on each boot?

Yes, exactly.

> Plus I would need to save the state of BSEC to EEPROM before going into deep sleep and load the state after waking up each time, is that correct?

This depends on how/if your program keeps the BSEC state in memory while in (deep) sleep.

Can you share your code on github?

Kamairo, 2019-01-20:
Thanks for the answers. I don't have any code regarding the BME680 yet as I am still trying to figure out what I need to make different than with the bme280. I used to work with the bme280 for my sensors and for that it's pretty straight forward 1. Set bme280 into forced mode 2. get readings (using the adafruit lib) 3. connect to wifi 4. publish the readings on my MQTT server 5. go into deep sleep for 25 min 6. repeat

cyberman54, 2019-01-20:
You can do so for BME680, too. It's not necessary to use the BSEC library. You can just read the sensor values with the BME680 API and do your own calculations without using the BSEC library.

Kamairo, 2019-01-20:
I know, but I thought Bosch isn't releasing their way of calculating the IAQ?!

cyberman54, 2019-01-20:
@Kamairo That's right. I you want to use their secret sauce you need to link the bsec library.

BoschSensortec, 2019-02-01:
Hi @Kamairo,

To the remaining questions (thanks @cyberman54 )

> What*s the difference between setting a configuration via bsec_set_configuration() and setting a sample rate via bsec_update_subscription()? I noticed that even when I load a 300s config, I can [...]

> What does exactly change between the 3.3V and the 1.8V configs? I mean, I can't change the voltage by software?! So what happens if I supply 3.3V to the BME and set the config to 1.8V and vice versa? 

A configuration is used to describe the configuration of the system. Right now limited to,

- Operation voltage : The LDO voltage regulator within the sensor generates heat when powered. The amount of heat generated depends on the applied VDD voltage. This is factored into the sensor models within the BSEC library. 

- Sampling duration : The 2 sampling modes of the sensor perform differently (reaction to the atmosphere) and it is strongly recommended to not switch between power modes

- Days in operation : This is kind of a jump start into the age of the sensor. There are 2 major checkpoints based on our tests, 4days and 28 days. After 28 days, all sensors tend to converge to similar behavior.