boschsensortec/BSEC-Arduino-library #46: multiple I2C devices problem

tedani, 2019-01-25:
I have a DS1307 RTC, 24C32 EEPROM and BME680 running on the same I2C bus, it seems the bsec.begin(i2c) class member is causing problem.  After calling bsec.begin(i2c), both the DS1307 and BME680 do not work.  

I found a temporary solution, in setup() function I need to call ds1307.begin()  followed by an operation to do with DS1307 before calling bsec.begin().  And, then, both DS1307 and BME680 will work normally.

Has anyone encountered such issue before?

ds1307.ino
`// Date and time functions using a DS1307 RTC connected via I2C and Wire lib
#include <NTPClient.h>
// change next line to use with another board/shield
#include <ESP8266WiFi.h>
//#include <WiFi.h> // for WiFi shield
//#include <WiFi101.h> // for WiFi 101 shield or MKR1000
#include <WiFiUdp.h>
#include <Wire.h>
#include "RTClib.h"
#include "bme680Sensor.h"

WiFiUDP ntpUDP;
NTPClient timeClient(ntpUDP, "uk.pool.ntp.org", 3600, 60000);

RTC_DS1307 rtc;
bme680Sensor iaqsensor;

char daysOfTheWeek[7][12] = {"Sunday", "Monday", "Tuesday", "Wednesday", "Thursday", "Friday", "Saturday"};
int rtc_running = 1, adjusted = 0;

void GetNetworkTime(void)
{
    String str;
    DateTime curtime;
    unsigned long t;
    uint16_t year;
    uint8_t month, day, hour, minute, second, dow;

    if(timeClient.update()) {
        Serial.println("NTP time:");
        t = timeClient.getEpochTime();
        str = "Epoch time:" + String(t) + String(" ===>>>  ");
        Serial.print(str);
        curtime = t;

        hour = curtime.hour();
        if (hour < 10)
            str = String('0') + String(hour);
        else
            str = String(hour);

        str += String (':');
        minute = curtime.minute();
        if (minute < 10)
            str += String('0') + String(minute);
        else
            str += String(minute);

        str += String (':');
        second = curtime.second();
        if (second < 10)
            str += String('0') + String(second);
        else
            str += String(second);

        str += String ("    ");

        day = curtime.day();
        str += String(day) + String('/');

        month = curtime.month();
        str += String(month) + String('/');

        year = curtime.year();
        str += String(year) + " (";

        dow = curtime.dayOfTheWeek();
        str += String(daysOfTheWeek[dow]);
        str += String(')');
        Serial.println(str);

        if (rtc_running && !adjusted) {
            rtc.adjust(curtime);
            adjusted = 1;
            }
        }
}

void UpdateIAQ(void)
{
    float temp;
    int humid, pressure;
    uint16_t iaq;
    uint8_t iaq_acc;

    if (iaqsensor.Read(&temp, &humid, &iaq, &iaq_acc, &pressure) == 1) {
        Serial.println("Temperature = " + String(temp,1));
        Serial.println("Humidity = " + String(humid));
        Serial.println(" IAQ = " + String(iaq));
        }
}

void setup ()
{
    char str[10];

    // Enable 3.3V - 5V TXS0104E voltage translator
    pinMode(D7, OUTPUT);
    digitalWrite(D7, HIGH);

    Serial.begin(115200);

    rtc.begin();

    WiFi.begin("myssid", "mypasswd");

    while ( WiFi.status() != WL_CONNECTED ) {
        delay ( 500 );
        Serial.print ( "." );
        }

    Serial.println("");
    timeClient.begin();

    rtc.writenvram(DS1307_NVRAM, (uint8_t*) "PaTtErN", 8);
    rtc.readnvram((uint8_t*) str, 8, DS1307_NVRAM);
    Serial.print("WR DS1307 cycle received: ");
    Serial.println(str);

    if (strcmp(str, "PaTtErN")) {
        rtc_running = 0;
        Serial.println("RTC is NOT running!");
        }

    iaqsensor.Begin();
}

void loop ()
{
    if (rtc_running) {
        DateTime now = rtc.now();
        Serial.print("RTC time: ");
        Serial.print(now.hour(), DEC);
        Serial.print(':');
        Serial.print(now.minute(), DEC);
        Serial.print(':');
        Serial.print(now.second(), DEC);
        Serial.print("   ");
        Serial.print(now.day(), DEC);
        Serial.print('/');
        Serial.print(now.month(), DEC);
        Serial.print('/');
        Serial.print(now.year(), DEC);
        Serial.print(" (");
        Serial.print(daysOfTheWeek[now.dayOfTheWeek()]);
        Serial.println(')');
        }

    GetNetworkTime();

    UpdateIAQ();

    delay(3000);
}`

My bme680Sensor.h
`#ifndef BME680SENSOR_H
#define BME680SENSOR_H

#include "bsec.h"
#include "bsec_serialized_configurations_iaq.h"

#define STATE_SAVE_PERIOD	UINT32_C(360 * 60 * 1000) // 360 minutes - 4 times a day

class bme680Sensor:public Bsec
{
    public:
        bme680Sensor():Bsec() {};
        int Begin(void);
        int Read(float *temp, int *humid, uint16_t *iaq, uint8_t *iaq_acc, int *pressure);
        void ErrorCode(int *bsecErr, int *bme680Err);
      	void setTemperatureOffset(float tempOffset);

    private:
        int bsecErrCode, bme680ErrCode;
        uint8_t bsecState[BSEC_MAX_STATE_BLOB_SIZE] = {0};
        uint16_t stateUpdateCounter = 0;

        int checkIaqSensorStatus(void);
        int loadState(void);
        int updateState(void);
};

#endif // BME680SENSOR_H`

My bme680Sensor.cpp
`#include <EEPROM.h>

#include "bme680Sensor.h"

#define STATE_SAVE_PERIOD	UINT32_C(360 * 60 * 1000) // 360 minutes - 4 times a day

int bme680Sensor::Begin(void)
{
    int ret = 0;

    bsec_virtual_sensor_t sensorList[7] = {
        BSEC_OUTPUT_RAW_TEMPERATURE,
        BSEC_OUTPUT_RAW_PRESSURE,
        BSEC_OUTPUT_RAW_HUMIDITY,
        BSEC_OUTPUT_RAW_GAS,
        BSEC_OUTPUT_IAQ,
        BSEC_OUTPUT_SENSOR_HEAT_COMPENSATED_TEMPERATURE,
        BSEC_OUTPUT_SENSOR_HEAT_COMPENSATED_HUMIDITY,
        };

    EEPROM.begin(BSEC_MAX_STATE_BLOB_SIZE + 1); // 1st address for the length

    Bsec::begin(BME680_I2C_ADDR_PRIMARY, Wire);

    ret = bme680Sensor::checkIaqSensorStatus();
    if (!ret) {
        Bsec::setConfig(bsec_config_iaq);
        ret = bme680Sensor::checkIaqSensorStatus();
        if (!ret) {
            ret = bme680Sensor::loadState();
            if (!ret) {
                Bsec::updateSubscription(sensorList, 7, BSEC_SAMPLE_RATE_LP);
                ret = checkIaqSensorStatus();
                }
            }
        }

    return ret;
}

int bme680Sensor::Read(float *temp, int *humid, uint16_t *iaq, uint8_t *iaq_acc, int *pressure)
{
    int ret = 0;

    if (Bsec::run()) { // If new data is available
        ret = 1;
        *pressure = (int) Bsec::pressure;
        *iaq = (uint16_t) Bsec::iaqEstimate;
        *iaq_acc = Bsec::iaqAccuracy;
        *temp = Bsec::temperature;
        *humid = (int) Bsec::humidity;
        if (updateState())
            ret = -1;
        }
    else
        if(bme680Sensor::checkIaqSensorStatus())
            ret = -1;

    return ret;
}

void bme680Sensor::setTemperatureOffset(float tempOffset)
{
    Bsec::setTemperatureOffset(tempOffset);
}

void bme680Sensor::ErrorCode(int *bsecErr, int *bme680Err)
{
    *bsecErr = bsecErrCode;
    *bme680Err = bme680ErrCode;
}

int bme680Sensor::checkIaqSensorStatus(void)
{
    int ret = 0;

    bsecErrCode = Bsec::status;
    if (bsecErrCode != BSEC_OK)
        ret = -1;

    bme680ErrCode = Bsec::bme680Status;
    if (bme680ErrCode != BME680_OK)
        ret = -1;

    return ret;
}

int bme680Sensor::loadState(void)
{
    int ret = 0;

    if (EEPROM.read(0) == BSEC_MAX_STATE_BLOB_SIZE) {
        // Existing state in EEPROM
        for (uint8_t i = 0; i < BSEC_MAX_STATE_BLOB_SIZE; i++)
            bsecState[i] = EEPROM.read(i + 1);

        Bsec::setState(bsecState);
        ret = bme680Sensor::checkIaqSensorStatus();
        }
    else {
        // Erase the EEPROM with zeroes
        for (uint8_t i = 0; i < BSEC_MAX_STATE_BLOB_SIZE + 1; i++)
            EEPROM.write(i, 0);
        EEPROM.commit();
        }

    return ret;
}

int bme680Sensor::updateState(void)
{
    int ret = 0;
    bool update = false;

    if (stateUpdateCounter == 0) {
        /* First state update when IAQ accuracy is >= 3 */
        if (Bsec::iaqAccuracy >= 3) {
            update = true;
            stateUpdateCounter++;
            }
        } 
    else {
        /* Update every STATE_SAVE_PERIOD minutes */
        if ((stateUpdateCounter * STATE_SAVE_PERIOD) < millis()) {
            update = true;
            stateUpdateCounter++;
            }
        }

    if (update) {
        Bsec::getState(bsecState);
        ret = bme680Sensor::checkIaqSensorStatus();
        if(!ret) {
            // Writing state to EEPROM
            for (uint8_t i = 0; i < BSEC_MAX_STATE_BLOB_SIZE ; i++)
                EEPROM.write(i + 1, bsecState[i]);

            EEPROM.write(0, BSEC_MAX_STATE_BLOB_SIZE);
            EEPROM.commit();
            }
        }

    return ret;
}
`

tedani, 2019-01-25:
ds1307.ino

// Date and time functions using a DS1307 RTC connected via I2C and Wire lib
#include <NTPClient.h>
// change next line to use with another board/shield
#include <ESP8266WiFi.h>
//#include <WiFi.h> // for WiFi shield
//#include <WiFi101.h> // for WiFi 101 shield or MKR1000
#include <WiFiUdp.h>
#include <Wire.h>
#include "RTClib.h"
#include "bme680Sensor.h"

WiFiUDP ntpUDP;
NTPClient timeClient(ntpUDP, "uk.pool.ntp.org", 3600, 60000);

RTC_DS1307 rtc;
bme680Sensor iaqsensor;

char daysOfTheWeek[7][12] = {"Sunday", "Monday", "Tuesday", "Wednesday", "Thursday", "Friday", "Saturday"};
int rtc_running = 1, adjusted = 0;

void GetNetworkTime(void)
{
    String str;
    DateTime curtime;
    unsigned long t;
    uint16_t year;
    uint8_t month, day, hour, minute, second, dow;

    if(timeClient.update()) {
        Serial.println("NTP time:");
        t = timeClient.getEpochTime();
        str = "Epoch time:" + String(t) + String(" ===>>>  ");
        Serial.print(str);
        curtime = t;

        hour = curtime.hour();
        if (hour < 10)
            str = String('0') + String(hour);
        else
            str = String(hour);

        str += String (':');
        minute = curtime.minute();
        if (minute < 10)
            str += String('0') + String(minute);
        else
            str += String(minute);

        str += String (':');
        second = curtime.second();
        if (second < 10)
            str += String('0') + String(second);
        else
            str += String(second);

        str += String ("    ");

        day = curtime.day();
        str += String(day) + String('/');

        month = curtime.month();
        str += String(month) + String('/');

        year = curtime.year();
        str += String(year) + " (";

        dow = curtime.dayOfTheWeek();
        str += String(daysOfTheWeek[dow]);
        str += String(')');
        Serial.println(str);

        if (rtc_running && !adjusted) {
            rtc.adjust(curtime);
            adjusted = 1;
            }
        }
}

void UpdateIAQ(void)
{
    float temp;
    int humid, pressure;
    uint16_t iaq;
    uint8_t iaq_acc;

    if (iaqsensor.Read(&temp, &humid, &iaq, &iaq_acc, &pressure) == 1) {
        Serial.println("Temperature = " + String(temp,1));
        Serial.println("Humidity = " + String(humid));
        Serial.println(" IAQ = " + String(iaq));
        }
}

void setup ()
{
    char str[10];

    // Enable 3.3V - 5V TXS0104E voltage translator
    pinMode(D7, OUTPUT);
    digitalWrite(D7, HIGH);

    Serial.begin(115200);

    rtc.begin();

    WiFi.begin("myssid", "mypasswd");

    while ( WiFi.status() != WL_CONNECTED ) {
        delay ( 500 );
        Serial.print ( "." );
        }

    Serial.println("");
    timeClient.begin();

    rtc.writenvram(DS1307_NVRAM, (uint8_t*) "PaTtErN", 8);
    rtc.readnvram((uint8_t*) str, 8, DS1307_NVRAM);
    Serial.print("WR DS1307 cycle received: ");
    Serial.println(str);

    if (strcmp(str, "PaTtErN")) {
        rtc_running = 0;
        Serial.println("RTC is NOT running!");
        }

    iaqsensor.Begin();
}

void loop ()
{
    if (rtc_running) {
        DateTime now = rtc.now();
        Serial.print("RTC time: ");
        Serial.print(now.hour(), DEC);
        Serial.print(':');
        Serial.print(now.minute(), DEC);
        Serial.print(':');
        Serial.print(now.second(), DEC);
        Serial.print("   ");
        Serial.print(now.day(), DEC);
        Serial.print('/');
        Serial.print(now.month(), DEC);
        Serial.print('/');
        Serial.print(now.year(), DEC);
        Serial.print(" (");
        Serial.print(daysOfTheWeek[now.dayOfTheWeek()]);
        Serial.println(')');
        }

    GetNetworkTime();

    UpdateIAQ();

    delay(3000);
}

tedani, 2019-01-25:
My bme680Sensor.h

#ifndef BME680SENSOR_H
#define BME680SENSOR_H

#include "bsec.h"
#include "bsec_serialized_configurations_iaq.h"

#define STATE_SAVE_PERIOD	UINT32_C(360 * 60 * 1000) // 360 minutes - 4 times a day

class bme680Sensor:public Bsec
{
    public:
        bme680Sensor():Bsec() {};
        int Begin(void);
        int Read(float *temp, int *humid, uint16_t *iaq, uint8_t *iaq_acc, int *pressure);
        void ErrorCode(int *bsecErr, int *bme680Err);
      	void setTemperatureOffset(float tempOffset);

    private:
        int bsecErrCode, bme680ErrCode;
        uint8_t bsecState[BSEC_MAX_STATE_BLOB_SIZE] = {0};
        uint16_t stateUpdateCounter = 0;

        int checkIaqSensorStatus(void);
        int loadState(void);
        int updateState(void);
};

#endif // BME680SENSOR_H

tedani, 2019-01-25:
My bme680Sensor.cpp

#include <EEPROM.h>

#include "bme680Sensor.h"

#define STATE_SAVE_PERIOD	UINT32_C(360 * 60 * 1000) // 360 minutes - 4 times a day

int bme680Sensor::Begin(void)
{
    int ret = 0;

    bsec_virtual_sensor_t sensorList[7] = {
        BSEC_OUTPUT_RAW_TEMPERATURE,
        BSEC_OUTPUT_RAW_PRESSURE,
        BSEC_OUTPUT_RAW_HUMIDITY,
        BSEC_OUTPUT_RAW_GAS,
        BSEC_OUTPUT_IAQ,
        BSEC_OUTPUT_SENSOR_HEAT_COMPENSATED_TEMPERATURE,
        BSEC_OUTPUT_SENSOR_HEAT_COMPENSATED_HUMIDITY,
        };

    EEPROM.begin(BSEC_MAX_STATE_BLOB_SIZE + 1); // 1st address for the length

    Bsec::begin(BME680_I2C_ADDR_PRIMARY, Wire);

    ret = bme680Sensor::checkIaqSensorStatus();
    if (!ret) {
        Bsec::setConfig(bsec_config_iaq);
        ret = bme680Sensor::checkIaqSensorStatus();
        if (!ret) {
            ret = bme680Sensor::loadState();
            if (!ret) {
                Bsec::updateSubscription(sensorList, 7, BSEC_SAMPLE_RATE_LP);
                ret = checkIaqSensorStatus();
                }
            }
        }

    return ret;
}

int bme680Sensor::Read(float *temp, int *humid, uint16_t *iaq, uint8_t *iaq_acc, int *pressure)
{
    int ret = 0;

    if (Bsec::run()) { // If new data is available
        ret = 1;
        *pressure = (int) Bsec::pressure;
        *iaq = (uint16_t) Bsec::iaqEstimate;
        *iaq_acc = Bsec::iaqAccuracy;
        *temp = Bsec::temperature;
        *humid = (int) Bsec::humidity;
        if (updateState())
            ret = -1;
        }
    else
        if(bme680Sensor::checkIaqSensorStatus())
            ret = -1;

    return ret;
}

void bme680Sensor::setTemperatureOffset(float tempOffset)
{
    Bsec::setTemperatureOffset(tempOffset);
}

void bme680Sensor::ErrorCode(int *bsecErr, int *bme680Err)
{
    *bsecErr = bsecErrCode;
    *bme680Err = bme680ErrCode;
}

int bme680Sensor::checkIaqSensorStatus(void)
{
    int ret = 0;

    bsecErrCode = Bsec::status;
    if (bsecErrCode != BSEC_OK)
        ret = -1;

    bme680ErrCode = Bsec::bme680Status;
    if (bme680ErrCode != BME680_OK)
        ret = -1;

    return ret;
}

int bme680Sensor::loadState(void)
{
    int ret = 0;

    if (EEPROM.read(0) == BSEC_MAX_STATE_BLOB_SIZE) {
        // Existing state in EEPROM
        for (uint8_t i = 0; i < BSEC_MAX_STATE_BLOB_SIZE; i++)
            bsecState[i] = EEPROM.read(i + 1);

        Bsec::setState(bsecState);
        ret = bme680Sensor::checkIaqSensorStatus();
        }
    else {
        // Erase the EEPROM with zeroes
        for (uint8_t i = 0; i < BSEC_MAX_STATE_BLOB_SIZE + 1; i++)
            EEPROM.write(i, 0);
        EEPROM.commit();
        }

    return ret;
}

int bme680Sensor::updateState(void)
{
    int ret = 0;
    bool update = false;

    if (stateUpdateCounter == 0) {
        /* First state update when IAQ accuracy is >= 3 */
        if (Bsec::iaqAccuracy >= 3) {
            update = true;
            stateUpdateCounter++;
            }
        } 
    else {
        /* Update every STATE_SAVE_PERIOD minutes */
        if ((stateUpdateCounter * STATE_SAVE_PERIOD) < millis()) {
            update = true;
            stateUpdateCounter++;
            }
        }

    if (update) {
        Bsec::getState(bsecState);
        ret = bme680Sensor::checkIaqSensorStatus();
        if(!ret) {
            // Writing state to EEPROM
            for (uint8_t i = 0; i < BSEC_MAX_STATE_BLOB_SIZE ; i++)
                EEPROM.write(i + 1, bsecState[i]);

            EEPROM.write(0, BSEC_MAX_STATE_BLOB_SIZE);
            EEPROM.commit();
            }
        }

    return ret;
}

cyberman54, 2019-01-25:
You could try using a mutex semaphore to ensure that no race conditions in i2c bus occur.

tedani, 2019-01-25:
> You could try using a mutex semaphore to ensure that no race conditions in i2c bus occur.

bsec uses TwoWire object while DS1307 uses Wire, would there be any conflicts?

cyberman54, 2019-01-25:
set log level to debug, watch for i2c bus errors and post (only) them here.

cyberman54, 2019-01-25:
@tedani `Wire` and `TwoWire` should be the same on 8266 arduino framework, see:
https://github.com/esp8266/Arduino/blob/22bab5f09bfe51ab0eb2bf0dd95ad617fc183a2f/libraries/Wire/Wire.cpp#L269